opentx/opentx #5073: How to reset radio to factory default parameter on the radio?

Jasonsiu, 2017-07-02:
How to reset radio to factory default parameter on the radio?

projectkk2glider, 2017-07-24:
If you are talking about OpenTX 2.2 on 9X radio then currently there is on such function.

Question to other devs @bsongis @kilrah @3djc @schwabe : why is this not enabled for non-arm platforms:

```c
void menuRadioVersion(event_t event)
{
   // some stuff...

#elif defined(CPUARM) && defined(EEPROM_RLC)
  lcdDrawTextAlignedLeft(MENU_HEADER_HEIGHT+5*FH+1, STR_EEBACKUP);
  lcdDrawTextAlignedLeft(MENU_HEADER_HEIGHT+6*FH+1, STR_FACTORYRESET);
  lcdDrawFilledRect(0, MENU_HEADER_HEIGHT+5*FH, LCD_W, 2*FH+1, SOLID);
  
  if (event == EVT_KEY_LONG(KEY_ENTER)) {
    eepromBackup();
  }
  else if (event == EVT_KEY_LONG(KEY_MENU)) {
    POPUP_CONFIRMATION(STR_CONFIRMRESET);
  }
#endif

}
```

The backup function obviously can't work, but factory reset could.

kilrah, 2017-07-24:
Flash usage?

schwabe, 2017-07-24:
I would just guess that it was implemented together with backup and never tested on AVR?

kilrah, 2017-07-24:
Yep. Pretty much nothing new has been ported to/tested on AVR for years...

closed 2017-11-08 by kilrah