opentx/opentx #5076: Companion 2.2.0 does not start

RustyPaint, 2017-07-02:
Running from the shortcut, nothing starts, and no error messages are given.

Running it in vs2015 community edition debugger gives this:

Exception thrown at 0x66AF9E46 (Qt5Core.dll) in companion.exe: 0xC0000005: Access violation reading location 0x05000004.

This is running on Windows 10 with all updates installed.

RustyPaint, 2017-07-02:
Downloading the QT SDK and copying over the DLLs used (and PDBs) shows that it is throwing the excption in:

>	Qt5Core.dll!QSettings::value(const QString & key, const QVariant & defaultValue) Line 3305	C++

RustyPaint, 2017-07-03:
Well... managed to build a debug build of Companion on my system.  Debug ran just fine.  And now the production version starts up just fine.

Airheadbit, 2017-07-13:
This sucks, mine fails on Qt5Core.dll

Airheadbit, 2017-07-13:
ANd then I find all the issues are caused by crap left in the windows registry by the 'Uninstaller'

projectkk2glider, 2017-07-24:
This might be a QT bug.

kilrah, 2017-07-24:
I think these crashes originate from a problem when importing settings from previous versions.

RustyPaint, 2017-07-25:
The previous version on my system is 2.0.19

mpaperno, 2017-11-12:
Duplicate of #5016, fixed a while ago in 655e2f9.

closed 2017-11-12 by mpaperno