opentx/opentx #5080: OpenTX Companion log viewer chooses incorrect icon path in google earth

Jmelz, 2017-07-06:
file://C:/Users/user_name/AppData/Local/Temp/15160-track0.png

Since the path has file:// appended to it, it causes google earth to display a red X as an icon instead of the more common arrow icon.

projectkk2glider, 2017-07-24:
The problem https://stackoverflow.com/a/24327346:
```
you could refer to the image by its absolute location (e.g. file:///C:/etnasss.jpg)
but Google Earth has security policy regarding access to local files on the file 
system outside the context of the KML file. You'd have to allow access to local 
files which generally is not recommended.
```

The solution is also described in the same answer.

Jmelz, 2017-07-24:
I think the best solution is to use the built-in google earth icons, but I'm not sure how that is implemented.

PR #5937 (2018-06-03): [Companion] Log viewer Google Earth fixes
  * Use relative path for Google Earth track icon (fixes #5080); 
  * Use detached process to launch GE (does not restart GE for every export, does not keep handle to GE process ([Qt docs](http://doc.qt.io/qt-5/qprocess.html#startDetached-1))); 
  * Fix typo in warning message;
  * Fix auto-connection warning from Qt due to misnamed function.
commit 97966548 (2018-06-03): [Companion] Log viewer Google Earth fixes (#5937)

* [Companion][LogsDialog] Use relative path for Google Earth track icon (fixes #5080); Use detached process to launch GE (does not restart GE for every export, does not keep handle to GE process); Fix typo in warning message.

* [LogsDialog] Use GE export file path as base for icon path (don't assume TempDir); Fix auto-connection warning from Qt due to misnamed function; Remove redundant check for existing GE file.
closed 2018-06-03 by bsongis