opentx/opentx #5082: RX8R sbus issue

Aris2024, 2017-07-07:
Transmitter: Horus x12s 2.2.0 218 EU version
RX 2x RX8R LBT 170208 (on a helicopter)

Master RX is bind in D16  telemetry mode, slave is bind in D16 without telemetry mode 

Master is connected to a skookum sk720 through sbus. Skookum has only the option to use futaba sbus protocol but I guess, that should be the same with the frsky sbus; When the system operates without the slave RX connected everything seems to be smooth and nice. When the slave receiver is connected to the RX in port of the master RX then the swash plate starts jittering up and down and there are a lot of uncommanded inputs. 
Especially in pitch aileron, elevator. 
The ppm servo ports are operating normally with out issue. 

If during the jittering I turn off the internal TX module then it immediately stops and comes back when I switch it on again. 
If the slave RX8R is replaced with an x8r, excactly the same is happening.
On a side note is it possible to update the firmware of the internal TX module of the horus and if yes how? 

Any clues why is this happening? Your help will be much appreciated.

kilrah, 2017-07-07:
You must ask this to FrSky or other users of the setup on forums, we do not control anything that happens receiver-side nor do we know about all the workings of that.

 The only thing we can recommend is you update firmware of the receivers and module (flash FrSky OS, boot once, see module being flashed, flash OpenTX again).

closed 2017-07-07 by kilrah

Aris2024, 2017-07-07:
ok thank you very much!