opentx/opentx #5083: Reassign sensors to inputs, mixes, special functions

lshems, 2017-07-07:
The current way sensors are used is confusing and cumbersome. If I bring to another receiver and back, I have to rediscover the sensors. Fine with that. 

But even when they are automagically named the same as before, there is no guarantee that previous assigned sensors still work on inputs, mixes and others stuff.

So you do not only have to rediscover every time, but also very carefully check all uses of sensors.

Thus had me nearly crash a plane, since I only use rssi, rcbatt and alt, and they are nicely rediscovered with the same name, suggesting they work as before.

kilrah, 2017-07-07:
Of course not, the name isn't meaningful at all, it's just for display. Index in the table is what counts.

But you do not need to rediscover when changing receiver with the same sensor config.

lshems, 2017-07-07:
D and X do have a different setup. So really confusing.

Also, after rediscovery, assignments do get lost.

But ok. I will live with it. Though it still seems dangerous to me to give
the impression things have been recognised by giving names. It surely
doesn't invite to check sensor settings after discovery, if you have been
restoring or copying models.

projectkk2glider, 2017-07-24:
> Though it still seems dangerous to me to give
the impression things have been recognized by giving names. It surely
doesn't invite to check sensor settings after discovery, if you have been
restoring or copying models.

I slightly agree with this. Do you have any good suggestion how this should be conveyed to the unsuspecting user?

schwabe, 2017-07-24:
Maybe when deleting a sensor/all sensors, check if a sensor is used somewhere and then warn them that a new sensor with the position will take over on rediscover?

geofrancis, 2017-07-25:
Why not keep the assignments in logical switches/special functions even after sensors are deleted

kilrah, 2017-07-25:
As mentioned earlier the references are broken since discovery order is random, no way for us to know what you intend to be what

geofrancis, 2017-07-25:
yes i mean just keep the name in the logical switches but add a star or something after the name it to highlight that the sensor is no longer what it expects. so you know what the sensor should be rather than what random sensor it has assigned.

kilrah, 2017-07-25:
Because the switches don't know the name. They only know they refer to "sensor 1", then for displaying sensor 1's name is called up. When the sensor is deleted the name is deleted too.

geofrancis, 2017-07-25:
is there no uniqe value that can be read from the sensors like smartport id or is that handled by the xjt module?

geofrancis, 2017-07-25:
is there no unique value that can be read from the sensors like smartport id or is that handled by the xjt module?

kilrah, 2017-07-25:
Of course and it is. And it is stored in the sensor data. And it is lost when the sensor is deleted.
The system is too flexible to be able to take guesses. Some people have 4 current sensors, which one are you using where?

geofrancis, 2017-07-25:
if opentx  it sorts the sensors using that unique value rather than the order of discovery so if you deleted all your sensors then rediscovered them again it would always go into the same order as long as the number of sensors has not changed.

kilrah, 2017-07-25:
That would break every setup where you relaunch discovery later to say just add one new sensor. And again can't work on setups with multiple of the same sensor.

geofrancis, 2017-07-25:
you could still add single sensors but yes you couldnt refresh all of them without changing the order.  but it would still work for multiples of the same sensor, it would still get every other sensor correct and you would just be left to sort those 2 or 3 sensors that have matching ids.

kilrah, 2017-07-25:
> currently every setup is currently broken anytime a refresh is done

But the point is you should almost never need to do a "refresh"! 

> even things like copying a model is made difficult, even if its an identical machine with identical receiver and flight controller putting out identical sensors it still ends up totally random 

Nope it should match perfectly.

That's the whole point, you're trying to find unsuitable solutions to a problem that doesn't exist in the first place if the system is used correctly.

geofrancis, 2017-07-25:
how can random be a good way of doing anything? all im saying is make some order to the sensors, doesnt matter if its frsky id or even alphabetically. just so long as they get ordered the same each time.

schwabe, 2017-07-25:
The order is not random, it only seems random, it is actually the order in which the sensors are discovered. As @kilrah explained your flaw is to delete all sensors. 

And even your approach has flaws. GPS sensor for example only start sending (all) telemetry after getting a fix.

geofrancis, 2017-07-25:
yes i know its the order is discovery but that is totally random.

flarssen, 2017-08-03:
#3354 would make it easier to have your own standard, but will of course lead to problems if users, unaware of the connection, rearrange the order after references are made.
What if all references are based on a combination of physical-, data- and group ID? Then the list order can be anything, perhaps also based on these ID's in that order, and alerts can be given if a referenced sensor is missing.

geofrancis, 2017-08-03:
opentx 2.0 was actually much easier to set up with sensors, since they didnt need discovered they would always be in the same order.

kilrah, 2017-08-03:
> What if all references are based on a combination of physical-, data- and group ID?

Again that would break whenever you discover extra sensors after initial discovery.

geofrancis, 2017-08-03:
it would only shift the sensors by 1 position if you added another sensor, when a new sensor is discovered and added to the list it could check if there are references to any moved sensor and change them accordingly.

flarssen, 2017-08-03:
> Again that would break whenever you discover extra sensors after initial discovery.
If my voltage sensor is referenced (in display, LS etc.) as 030210 instead of index 4, I can't see how an extra sensor could break this. 
Not sure how it would work out performance wise. Then there is calculated sensors which doesn't have these ID's.

kilrah, 2017-08-04:
> A simpler solution would be if it just ordered them for initial discovery, then add sesors just put them on the end of the list like it does currently. it would mean the sensors would be always in the same order apart from any single sensor that was not part of the setup initially.

Then it's completely pointless, you'll end up in the same situation as now most of the time - except worse since it's not consistent and you never know what to expect.

> If my voltage sensor is referenced (in display, LS etc.) as 030210 instead of index 4,

That can't happen without huge changes to the way everything works in OpenTX - so it won't. 

Once again when you make a copy of a setup as seems to be the main reason for complaining everything should work as is out of the box, so again we're losing time on a complete non-issue. You have both conveniently ignored that argument.

lshems, 2020-01-08:
Still relevant, and reports on crashes because of it have in the meantime become available.

closed 2020-10-04 by stale[bot]