opentx/opentx #5086: OpenTX 2.2 range issues.

qemay, 2017-07-10:
Hello,

I have recently updated my Taranias x9d to the 2.2 branch (lua, noheli, non eu) from the 2.1.7. The update itself was easy, but I now experience a extreme range problem with my XSR.

The transmitter acts as it would be in a constant range check mode if not worse. I get a low warning at 20m, critical at 30m and a few failsafe switches in this range. The exact same receiver was working fine before the update (at least few hundred meters range).

I will try to switch the receiver, but could it be related to the update?

3djc, 2017-07-10:
Opentx does not deal with the radio link itself, so no reason for it to be related

closed 2017-07-10 by 3djc

qemay, 2017-07-10:
Is it not? The radio link (its protocol) is configured from within OpenTX, the range check is configured from within OpenTX. 
So it's easy to say.. hardware issue not my problem. So the update broke my hardware as it was perfectly working before the update.
Case closed. Thank you for your help.

kilrah, 2017-07-10:
Of course OpenTX tells the module what to do... but if it was telling the module to be in range check mode don't you think everyone would have the same problem?