opentx/opentx #5087: Stickrev broken again

zmoothrc, 2017-07-10:
Just updated my 9X to 2.2.0 from 2.0 and stickrev stopped working. Missing from compile or bug?

projectkk2glider, 2017-07-24:
'FRSKY_STICKS' option is missing from the cmake files! 

It is present in the compilation script.

projectkk2glider, 2017-07-24:
Another thing to check. The `"stickrev": ("FRSKY_STICKS", "YES", "NO")` is present in the `fwoptions.py` for the 9X, but not for the 9XR, while the source suggests that 9XR also supports this option.

kilrah, 2017-07-24:
I believe that was intentional becasue FrSky sticks don't physically fit in the 9XR.

kilrah, 2017-07-24:
Hmm, it is available for selection in companion, so maybe not

3djc, 2017-08-17:
```
elif options[optcount] == "9xr":
    command_options["PCB"] = "9XR"
    firmware_options = options_9x
```
So I'm guessing 9XR is covered as well

PR #5177 (2017-08-22): FRSKY_STICKS option was missing in 2.2
This fixes #5087
closed 2017-08-22 by bsongis