opentx/opentx #5090: Some sensors created with telemetry simulator do not update

jtaylor2, 2017-07-14:
Discovered while looking into problem reported by rcyoda on  http://openrcforums.com/forum/viewtopic.php?f=94&t=10355&p=130160#p130129

Tested with Companion2.2 x9d simulator
If a rxbat, A1, or A2 sensor is created by using the telemetry simulator it will not update.  Steps to recreate:

1. In Companion create or simulate a model with no sensors detected
2. Open telemetry simulator and add values for rxbt, A1, or A2
3. On radio simulator go to telemetry screen and "discover new sensors"
4. Sensors are discovered, but do not update.

  The reason for this is that the telemetry simulators ratio field is read and set from the existing sensors.  If there are no sensors it is set to 0.0.  This field is not editable so the sensor is created but will not update (shows 0).  Even if the ratio is changed on the radio simulator it will not update because the ratio on the telemetry simulator is still 0 and not editable.

I will create a pull request to allow the telemetry simulator rxbt, A1, and A2 ratio fields be editable.  This will allow the user to change those fields and add a ratio that will then have the sensor update.

Question:  Is this enough or should I add default ratios for the rxbat, A1, and A2 sensors and if so what should the defaults be.

kilrah, 2017-07-14:
> I will create a pull request to allow the telemetry simulator rxbt, A1, and A2 ratio fields be editable.

That would not be appropriate unless you also introduced a mechanism to synchronise ratios between the simulator and the radio. That may not be easy, which is why the value is currently locked for editing (it's only read from the eeprom on simulator load).

jtaylor2, 2017-07-14:
@kilrah I guess I'm not understanding something.  I thought the telemetry simulator was used for testing and simulation only and changes made in the simulator never got written to the radio, or in the case of the Companion simulator, not even saved on exit.  If the sensor exists in the eeprom being simulated the telemetry simulator uses the ratio read from the eeprom, but if the sensor does not exist the ratio is set to 0 and discovery creates a new sensor which does not update because the ratio is 0  and can't be changed in the telemetry simulator.  

The sensor is still created, but doesn't update.  It's not going to be written to the radio anyway (and if it was it would have a bad ratio) so I don't understand why making the telemetry simulator ratio editable to make the simulation work would hurt anything.

3djc, 2017-07-15:
Companion simulator doesn't change your otx, but the standalone simulator does

kilrah, 2017-07-15:
> The sensor is still created, but doesn't update. It's not going to be written to the radio anyway (and if it was it would have a bad ratio) so I don't understand why making the telemetry simulator ratio editable to make the simulation work would hurt anything.

The point is it won't make the simulation work. The ratio you enter in the telemtry simulator will never make its way into the (virtual) radio. If there was no sensor in the eeprom and you discover it within the simulator it will be created with the default of 13.2. There is no mechanism to convey the ratio set in the telemetry simu to the radio.

jtaylor2, 2017-07-15:
> The point is it won't make the simulation work as you intend. The ratio you enter in the telemetry simulator will never make its way into the (virtual) radio.

Yes, I understand that.  For a new sensor the user must set the ratio the same in the simulator and the (virtual) radio.   Messy but if you can edit the ratio in the simulator at least it gives a way to make it work.  My other thought was if the sensor did not exist to set the simulator ratio to the default (13.2), but without the field being editable it would still break (as it does now) if the user changed the ratio on the ratio.

Rewriting the simulator for two way communications is far beyond my ability.  The best I could offer was a hack to make it work.  I still don't understand why being able to edit the ratio field is more harmful than being able to edit the instance,  In any case I'll delete my pull request and unassign myself from this issue.

schwabe, 2017-07-15:
Users know two things:

* Things in the simulator are not being saved
* Things done outside the simulator are saved.

So if you allow changing the ratio outside the simulator, users might assume that this is written back to the otx file.

kilrah, 2017-07-15:
> I still don't understand why being able to edit the ratio field is more harmful than being able to edit the instance

Wrong instance = no data. If the ratio can be edited you can be sure any normal user will expect the one in the radio to change too, and since it doesn't he'll get wrong data and not understand anything.

The solution is simple - Make sure your sensors are in there before using the telemetry simulator.

>  My other thought was if the sensor did not exist to set the simulator ratio to the default (13.2)

That is the only proper solution I had in mind at this point.

jtaylor2, 2017-07-15:
> The solution is simple - Make sure your sensors are in there before using the telemetry simulator. Discovering in the simulator is not really an intended way of operating.

But discovering in the simulator is how at least some people use it, as did the openrcforums user whose post got me looking at it.  People test sensor configs in the simulator before they have them or, as I did, attempt to simulator other peoples configs.

> If anything had to be done that is the only proper solution I had in mind at this point.

I actually tested adding a ratio default if the sensor doesn't exist and it is pretty easy to do, but without being able to edit the ratio it might not be what was wanted and still wouldn't solve the problem of the user changing the ratio in the radio and getting the wrong data.  I could do that, but not sure it would be any better than the existing problem where the sensor gets created but doesn't update.   Particularly since using the standalone simulator and closing and reopening it is a viable workaround.

Perhaps someone (Max) with the ability will do it right.

projectkk2glider, 2017-07-24:
The problem of the sensor showing 0.0V is somewhere else. I will look into it.

PR #5110 (2017-07-30): Fixes #5090: periodically update sensor ratios from the radio when th…
…e telemetry simulator is running.

Closes #5090
commit 3436273f (2017-07-30): Fixes #5090: periodically update sensor ratios from the radio when the telemetry simulator is running. (#5110)
closed 2017-07-30 by bsongis