opentx/opentx #5092: opentx 9x128 green ed040502-h16d invalid eprom filesystem

s3melle, 2017-07-14:
Hi,

I have an invalid eeprom when flashing with opentx.
The remote is warning after a flash and it's formating the eeprom.

Same probleme with companinon 2.1 and 2.2, he can't read the eeprom data is invalid, can't open radio.bin

Thanks

kilrah, 2017-07-15:
Please be clear on what you're actually trying to do - it seems you try to read the eeprom from the stock firmware before flashing OpenTX, and it's normal for that not to be readable.

s3melle, 2017-07-16:
I'm telling that after every falshing with opentx the rom is formating because it's invalid and if i create some profile i can 't export it with companion because it 's not redable.

projectkk2glider, 2017-07-24:
I think you have problems with `usbasp` same as #5093

s3melle, 2017-07-25:
For the #5092 - invalid eeprom, i'm not sure it's the same issue with the #5093 how only add an udev rules, i have the probleme under windows too.

drepou, 2017-07-27:
same issue for me impossible to writte the eeprom only compagnion 2.0 is working
after that issue with eeprom version.

I dont understand the uploaded version of FW 2.2.0 use a 216-2 eeprom version ( fw is named with 216-2 ) at the end

I try to force the epprom erase with avrdude it 's ok i get FF every where in epprom.
Then reflash the FW with compagnion get the invalid eeprom message on the radio and "erasing" message then the old eeprom is back again !! and error eeprom upgrade when reading.

I try to create a empty profile and save into the radio same error after ...

kilrah, 2017-11-08:
Likely mismatched companion and firmware options.

closed 2017-11-08 by kilrah

s3melle, 2018-02-11:
semelle@dobian:~$ companion22 
^[^[[15~^['^[([D] Translations::installTranslators():101 - Locale name: "fr_FR" language: "français" country: "France"
[D] Translations::tryLoadTranslator():158 - Installed translation file "qt" from "/usr/share/qt5/translations" for "fr_FR"
[D] Translations::tryLoadTranslator():158 - Installed translation file "qtbase" from "/usr/share/qt5/translations" for "fr_FR"
[D] Translations::tryLoadTranslator():158 - Installed translation file "companion" from ":/translations" for "fr_FR"
[D] registerStorageFactory():54 - register storage "bin"
[D] registerStorageFactory():54 - register storage "eepe"
[D] registerStorageFactory():54 - register storage "hex"
[D] registerStorageFactory():54 - register storage "otx"
[D] registerStorageFactory():54 - register storage "sdcard"
[D] SimulatorLoader::registerSimulators():55 - [simulator.interface.loader] Searching for simulators in "." matching pattern ("*-simulator.so")
[D] SimulatorLoader::registerSimulators():83 - [simulator.interface.loader] Found libraries: ("none")
[D] SimulatorLoader::registerSimulators():55 - [simulator.interface.loader] Searching for simulators in "/usr/local/lib/companion22" matching pattern ("*-simulator.so")
[D] SimulatorLoader::registerSimulators():60 - [simulator.interface.loader] Trying to register simulator in  "libopentx-9x-simulator.so"
[D] SimulatorLoader::registerSimulators():75 - [simulator.interface.loader] Registered "opentx-9x" simulator in  "/usr/local/lib/companion22/libopentx-9x-simulator.so" and unloaded: true
[D] SimulatorLoader::registerSimulators():60 - [simulator.interface.loader] Trying to register simulator in  "libopentx-9x128-simulator.so"
[D] SimulatorLoader::registerSimulators():75 - [simulator.interface.loader] Registered "opentx-9x128" simulator in  "/usr/local/lib/companion22/libopentx-9x128-simulator.so" and unloaded: true
[D] SimulatorLoader::registerSimulators():60 - [simulator.interface.loader] Trying to register simulator in  "libopentx-9xr-simulator.so"
[D] SimulatorLoader::registerSimulators():75 - [simulator.interface.loader] Registered "opentx-9xr" simulator in  "/usr/local/lib/companion22/libopentx-9xr-simulator.so" and unloaded: true
[D] SimulatorLoader::registerSimulators():60 - [simulator.interface.loader] Trying to register simulator in  "libopentx-9xr128-simulator.so"
[D] SimulatorLoader::registerSimulators():75 - [simulator.interface.loader] Registered "opentx-9xr128" simulator in  "/usr/local/lib/companion22/libopentx-9xr128-simulator.so" and unloaded: true
[D] SimulatorLoader::registerSimulators():60 - [simulator.interface.loader] Trying to register simulator in  "libopentx-9xrpro-simulator.so"
[D] SimulatorLoader::registerSimulators():75 - [simulator.interface.loader] Registered "opentx-9xrpro" simulator in  "/usr/local/lib/companion22/libopentx-9xrpro-simulator.so" and unloaded: true
[D] SimulatorLoader::registerSimulators():60 - [simulator.interface.loader] Trying to register simulator in  "libopentx-ar9x-simulator.so"
[D] SimulatorLoader::registerSimulators():75 - [simulator.interface.loader] Registered "opentx-ar9x" simulator in  "/usr/local/lib/companion22/libopentx-ar9x-simulator.so" and unloaded: true
[D] SimulatorLoader::registerSimulators():60 - [simulator.interface.loader] Trying to register simulator in  "libopentx-gruvin9x-simulator.so"
[D] SimulatorLoader::registerSimulators():75 - [simulator.interface.loader] Registered "opentx-gruvin9x" simulator in  "/usr/local/lib/companion22/libopentx-gruvin9x-simulator.so" and unloaded: true
[D] SimulatorLoader::registerSimulators():60 - [simulator.interface.loader] Trying to register simulator in  "libopentx-mega2560-simulator.so"
[D] SimulatorLoader::registerSimulators():75 - [simulator.interface.loader] Registered "opentx-mega2560" simulator in  "/usr/local/lib/companion22/libopentx-mega2560-simulator.so" and unloaded: true
[D] SimulatorLoader::registerSimulators():60 - [simulator.interface.loader] Trying to register simulator in  "libopentx-sky9x-simulator.so"
[D] SimulatorLoader::registerSimulators():75 - [simulator.interface.loader] Registered "opentx-sky9x" simulator in  "/usr/local/lib/companion22/libopentx-sky9x-simulator.so" and unloaded: true
[D] SimulatorLoader::registerSimulators():60 - [simulator.interface.loader] Trying to register simulator in  "libopentx-x10-simulator.so"
register theme Default
register theme Darkblue
register layout Layout2x2
register layout Layout1x1
register layout Layout2x1
register layout Layout2P1
register layout Layout2x4
register widget Text
register widget Timer
register widget ModelBmp
register widget Outputs
register widget Value
register widget Gauge
[D] SimulatorLoader::registerSimulators():75 - [simulator.interface.loader] Registered "opentx-x10" simulator in  "/usr/local/lib/companion22/libopentx-x10-simulator.so" and unloaded: true
[D] SimulatorLoader::registerSimulators():60 - [simulator.interface.loader] Trying to register simulator in  "libopentx-x12s-simulator.so"
register theme Default
register theme Darkblue
register layout Layout2x2
register layout Layout1x1
register layout Layout2x1
register layout Layout2P1
register layout Layout2x4
register widget Text
register widget Timer
register widget ModelBmp
register widget Outputs
register widget Value
register widget Gauge
[... cut ...]
[D] OpenTxEepromInterface::load():186 - trying OpenTX for M128 / 9X board import... version 216 ko
[W] unknown():0 - QXcbConnection: XCB error: 3 (BadWindow), sequence: 14909, resource id: 4195086, major code: 40 (TranslateCoords), minor code: 0
[D] OpenTxEepromInterface::load():186 - trying OpenTX for 9X board import... version 216 variant 0 ok
[W] unknown():0 - QXcbConnection: XCB error: 3 (BadWindow), sequence: 15025, resource id: 4195095, major code: 40 (TranslateCoords), minor code: 0
[D] main():165 - COMPANION EXIT 0