opentx/opentx #5094: PXX CRC Calculation - what is polynomial, and other params for calculation

krzysztofkuczek, 2017-07-14:
Hi,
I'm writing pxx decoded for my LRS, and trying to generate CRC calculation routines using pycrc..
I used polynomial 0x1189 but still crc calculated by my routine is different.
I can't use you codding because table is tooooo big to fit in little flash memory.
Can you provide more info how you generated your crc calculation routines.
Thank you in advance
Krzysiek

kilrah, 2017-07-15:
Sorry but we were given this table to use as is, we don't have any insight on how it was created.

The current doc we have does mention a polynomial of 0x1189.

closed 2017-07-15 by kilrah

schwabe, 2017-07-15:
if you google our table you will end up with this:

https://github.com/spotify/linux/blob/master/lib/crc-ccitt.c

```
 * This mysterious table is just the CRC of each possible byte. It can be
 * computed using the standard bit-at-a-time methods. The polynomial can
 * be seen in entry 128, 0x8408. This corresponds to x^0 + x^5 + x^12.
 * Add the implicit x^16, and you have the standard CRC-CCITT.
```

krzysztofkuczek, 2017-07-15:
Thx, will check it

krzysztofkuczek, 2017-07-17:
Hi,
I just wonder if you have everything ok with CRC calculation. I wrote my PXX decoder, and channels values are ok, but when i look into pxx frame, i can see that low byte of CRC sum is always ZERO.
Sample frames:
126,11,64,0,253,51,63,0,161,63,153,4,112,100,34,17,0,170,126,
126,11,64,0,253,35,63,0,161,63,153,4,112,100,34,17,0,59,126,

On the face of it, the signal also repentants zero at this position
[image]

According to limited documentation, and source code, there should be LO and HIGH byte of CRC sum..
Or maybe i do something wrong.
Krzysiek