opentx/opentx #5095: Rerunning make rebuilds opentx-*-simulator again.

adelton, 2017-07-15:
On Fedora 26, I build the 2.1 version of companion with

```
# dnf install -y git svn qt qt-devel cmake patch xsd gcc-c++ SDL-devel phonon phonon-devel
# dnf install -y xerces-c-devel PyQt4 python2 avr-gcc findutils
# git clone -b 2.1 https://github.com/opentx/opentx.git opentx
# mkdir opentx/companion/lbuild
# cd opentx/companion/lbuild
# cmake ../src
# make
```

The build passes, however, when I run `make` again, builds of

`firmwares/opentx/simulator/CMakeFiles/opentx-*-simulator.dir/opentxsimulator.cpp.o`

are run again:

```
# make
[  1%] Built target qcustomplot
[  6%] Built target common
[ 12%] Built target shared
[ 30%] Built target modeledit
[ 36%] Built target generaledit
[ 47%] Built target simulation
[ 87%] Built target companion21
[ 88%] Built target qxtcommandoptions
[ 91%] Built target simulator21
[ 91%] Built target about_lbm_taranis
[ 92%] Built target lua_export_taranis_x9e
[ 92%] Built target commondeps
Scanning dependencies of target opentx-taranisx9e-simulator
[ 92%] Building CXX object firmwares/opentx/simulator/CMakeFiles/opentx-taranisx9e-simulator.dir/opentxsimulator.cpp.o
[ 93%] Linking CXX shared library ../../../libopentx-taranisx9e-simulator.so
[ 93%] Built target opentx-taranisx9e-simulator
[ 93%] Built target about_lbm_9x
Scanning dependencies of target opentx-9x-simulator
[ 94%] Building CXX object firmwares/opentx/simulator/CMakeFiles/opentx-9x-simulator.dir/opentxsimulator.cpp.o
In file included from /opentx/companion/src/firmwares/opentx/simulator/opentxsimulator.cpp:114:0:
/opentx/companion/src/../../radio/src/opentx.cpp: In function 'uint8_t M64::checkTrim(uint8_t)':
/opentx/companion/src/../../radio/src/opentx.cpp:1352:31: warning: ?: using integer constants in boolean context, the expression will always evaluate to 'true' [-Wint-in-bool-context]
         beepTrim = (mark == 0 ? 1 : 2);
                    ~~~~~~~~~~~^~~~~~~~
[ 94%] Linking CXX shared library ../../../libopentx-9x-simulator.so
[ 94%] Built target opentx-9x-simulator
Scanning dependencies of target opentx-gruvin9x-simulator
[ 94%] Building CXX object firmwares/opentx/simulator/CMakeFiles/opentx-gruvin9x-simulator.dir/opentxsimulator.cpp.o
In file included from /opentx/companion/src/firmwares/opentx/simulator/opentxsimulator.cpp:114:0:
/opentx/companion/src/../../radio/src/opentx.cpp: In function 'uint8_t Gruvin9X::checkTrim(uint8_t)':
/opentx/companion/src/../../radio/src/opentx.cpp:1352:31: warning: ?: using integer constants in boolean context, the expression will always evaluate to 'true' [-Wint-in-bool-context]
         beepTrim = (mark == 0 ? 1 : 2);
                    ~~~~~~~~~~~^~~~~~~~
[ 95%] Linking CXX shared library ../../../libopentx-gruvin9x-simulator.so
[ 95%] Built target opentx-gruvin9x-simulator
Scanning dependencies of target opentx-9x128-simulator
[ 95%] Building CXX object firmwares/opentx/simulator/CMakeFiles/opentx-9x128-simulator.dir/opentxsimulator.cpp.o
In file included from /opentx/companion/src/firmwares/opentx/simulator/opentxsimulator.cpp:114:0:
/opentx/companion/src/../../radio/src/opentx.cpp: In function 'uint8_t M128::checkTrim(uint8_t)':
/opentx/companion/src/../../radio/src/opentx.cpp:1352:31: warning: ?: using integer constants in boolean context, the expression will always evaluate to 'true' [-Wint-in-bool-context]
         beepTrim = (mark == 0 ? 1 : 2);
                    ~~~~~~~~~~~^~~~~~~~
[ 95%] Linking CXX shared library ../../../libopentx-9x128-simulator.so
[ 95%] Built target opentx-9x128-simulator
Scanning dependencies of target opentx-9xr128-simulator
[ 95%] Building CXX object firmwares/opentx/simulator/CMakeFiles/opentx-9xr128-simulator.dir/opentxsimulator.cpp.o
In file included from /opentx/companion/src/firmwares/opentx/simulator/opentxsimulator.cpp:114:0:
/opentx/companion/src/../../radio/src/opentx.cpp: In function 'uint8_t M128XR::checkTrim(uint8_t)':
/opentx/companion/src/../../radio/src/opentx.cpp:1352:31: warning: ?: using integer constants in boolean context, the expression will always evaluate to 'true' [-Wint-in-bool-context]
         beepTrim = (mark == 0 ? 1 : 2);
                    ~~~~~~~~~~~^~~~~~~~
[ 96%] Linking CXX shared library ../../../libopentx-9xr128-simulator.so
[ 96%] Built target opentx-9xr128-simulator
[ 96%] Built target lua_export_taranis
Scanning dependencies of target opentx-taranisplus-simulator
[ 97%] Building CXX object firmwares/opentx/simulator/CMakeFiles/opentx-taranisplus-simulator.dir/opentxsimulator.cpp.o
[ 97%] Linking CXX shared library ../../../libopentx-taranisplus-simulator.so
[ 97%] Built target opentx-taranisplus-simulator
Scanning dependencies of target opentx-9xr-simulator
[ 98%] Building CXX object firmwares/opentx/simulator/CMakeFiles/opentx-9xr-simulator.dir/opentxsimulator.cpp.o
In file included from /opentx/companion/src/firmwares/opentx/simulator/opentxsimulator.cpp:114:0:
/opentx/companion/src/../../radio/src/opentx.cpp: In function 'uint8_t M64XR::checkTrim(uint8_t)':
/opentx/companion/src/../../radio/src/opentx.cpp:1352:31: warning: ?: using integer constants in boolean context, the expression will always evaluate to 'true' [-Wint-in-bool-context]
         beepTrim = (mark == 0 ? 1 : 2);
                    ~~~~~~~~~~~^~~~~~~~
[ 98%] Linking CXX shared library ../../../libopentx-9xr-simulator.so
[ 98%] Built target opentx-9xr-simulator
Scanning dependencies of target opentx-mega2560-simulator
[ 98%] Building CXX object firmwares/opentx/simulator/CMakeFiles/opentx-mega2560-simulator.dir/opentxsimulator.cpp.o
In file included from /opentx/companion/src/firmwares/opentx/simulator/opentxsimulator.cpp:114:0:
/opentx/companion/src/../../radio/src/opentx.cpp: In function 'uint8_t Mega2560::checkTrim(uint8_t)':
/opentx/companion/src/../../radio/src/opentx.cpp:1352:31: warning: ?: using integer constants in boolean context, the expression will always evaluate to 'true' [-Wint-in-bool-context]
         beepTrim = (mark == 0 ? 1 : 2);
                    ~~~~~~~~~~~^~~~~~~~
[ 98%] Linking CXX shared library ../../../libopentx-mega2560-simulator.so
[ 98%] Built target opentx-mega2560-simulator
Scanning dependencies of target opentx-sky9x-simulator
[ 99%] Building CXX object firmwares/opentx/simulator/CMakeFiles/opentx-sky9x-simulator.dir/opentxsimulator.cpp.o
[ 99%] Linking CXX shared library ../../../libopentx-sky9x-simulator.so
[ 99%] Built target opentx-sky9x-simulator
Scanning dependencies of target opentx-9xrpro-simulator
[ 99%] Building CXX object firmwares/opentx/simulator/CMakeFiles/opentx-9xrpro-simulator.dir/opentxsimulator.cpp.o
[ 99%] Linking CXX shared library ../../../libopentx-9xrpro-simulator.so
[ 99%] Built target opentx-9xrpro-simulator
Scanning dependencies of target opentx-ar9x-simulator
[100%] Building CXX object firmwares/opentx/simulator/CMakeFiles/opentx-ar9x-simulator.dir/opentxsimulator.cpp.o
[100%] Linking CXX shared library ../../../libopentx-ar9x-simulator.so
[100%] Built target opentx-ar9x-simulator
Scanning dependencies of target opentx-taranis-simulator
[100%] Building CXX object firmwares/opentx/simulator/CMakeFiles/opentx-taranis-simulator.dir/opentxsimulator.cpp.o
[100%] Linking CXX shared library ../../../libopentx-taranis-simulator.so
[100%] Built target opentx-taranis-simulator
```

The strange thing is that the files like `firmwares/opentx/simulator/CMakeFiles/opentx-taranis-simulator.dir/opentxsimulator.cpp.o` exist after `make` finishes yet running with `-d` says that it does not exist. So it looks like the file gets purged somehow in the process.

This is with

```
# cmake --version
cmake version 3.8.2

CMake suite maintained and supported by Kitware (kitware.com/cmake).
```
```
# make --version
GNU Make 4.2.1
Built for x86_64-redhat-linux-gnu
Copyright (C) 1988-2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
```

projectkk2glider, 2017-07-24:
Not sure we will even fix this. Version 2.2 uses much more elaborate cmake files and there this problem does not exist.

adelton, 2017-07-25:
Have you got any suggestion about what might be causing this, or how to workaround the issue, potentially by tweaking the `Makefile`s after `cmake` finished running?

schwabe, 2017-07-25:
@adelton why are trying to cling to 2.1 and are not upgrading to 2.2? It is unlikely that we will invest time in 2.1 now that 2.2 is out.

adelton, 2017-07-25:
Because 2.2 currently fails with segfault for me so when I was upgrading my Fedora copr repo https://copr.fedorainfracloud.org/coprs/adelton/opentx-companion/ from 2.0 to reasonable newer version, I went with 2.1.

kilrah, 2017-07-25:
Maybe giving reporting the issue with relevant data so 2.2 can be fixed instead makes more sense... We don't/didn't know about it until now.

3djc, 2017-07-25:
And unless critical bug are found, we are not planning to fix 2.1.x for smaller things. 2.2.x is where it happens !

adelton, 2017-07-25:
> Maybe reporting the issue with relevant data so 2.2 can be fixed instead makes more sense... We don't/didn't know about it until now.

I will report it when I find time to focus on it again some more. It might well have been my `.spec` file which was wrong when I tried it, and I like to do my homework before throwing issues around. That's why I went with 2.1 where the delta compared to 2.0 was smaller.

adelton, 2017-07-25:
> And unless critical bug are found, we are not planning to fix 2.1.x for smaller things. 2.2.x is where it happens !

I'm not asking for new release with a fix. I ask about opinions about the cause of the behaviour that I see.

schwabe, 2017-07-25:
@adelton We rewrote large parts of Cmake for 2.2, which is also why you are seeing a larger delta for 2.2 than for 2.1. Especially building the radio/simulator sources has been rewritten. Basically issues like you are seeing have been hopefully been addressed by this rewrite.

Also I don't see an error/real problem in your initial post other than annoyance with rebuilding the file over and over.

adelton, 2017-07-25:
> Also I don't see an error/real problem in your initial post other than annoyance with rebuilding the file over and over.

Yes, it's just an annoyance. But sometimes annoyances can be symptom of some deeper problem or bug. That's why I filed this issue.

schwabe, 2017-07-25:
@adelton I agree but in this case the annoyance should be fixed in the current code and 2.1 does only get criticial bugfixes.

adelton, 2017-08-20:
Just reporting that I was able to build the 2.2 version as rpm on Fedora:
https://copr.fedorainfracloud.org/coprs/adelton/opentx-companion-2.2/

closed 2017-08-20 by adelton