opentx/opentx #5097: [COMPANION]

stanislavdavid, 2017-07-19:
Hi, I am not able to open any model in actual version of companion. Nothing happens when i doubleclick any model.

Thanks for help

I am enclosing file

[aktualni.zip](https://github.com/opentx/opentx/files/1160282/aktualni.zip)

kilrah, 2017-07-19:
Radio type? 
What exact companion version?
What firmware version on the radio?
Where does this file come from, what's the history?

stanislavdavid, 2017-07-19:
I have Taranis X9D plus all latest firmwares 2.2.0. First model is based on default plane with flaps, slightly modified on radio. All others are dummy and example models from my reseller. I am not able to use in companion any of them

On my second PC i am able to open it without problems.

Thanks for help

kilrah, 2017-07-20:
Likely wrong radio type selected in companion. No issue here.

closed 2017-07-20 by kilrah