opentx/opentx #5102: FrSky S6R receiver bind problem

Alexander-C, 2017-07-21:
I have 3 S6R receivers and I have the same binding problem with all three.
I am using a Taranis Plus updated with the latest firmware Ver 170317 and the S6R receivers upated to the latest stable version 170217 (NON EU)
Companion Ver 2.2 and Open TX 2.2.0
Create a model on an empty slot. On some slots the receiver will refuse to bind but editing the model and manually changing the receiver number to the one lower than allocated when creating the model it will now bind.
For example I created new model on slot 15, model was then given a receiver number of 15 matching the model number. The receiver does not bind. Manually changing the receiver number to 14 allows the receiver to bind.
One thing I did notice was that the receiver numbers available on the transmitter range from 00 to 63 while on Companion it ranges from 01 to 63
As this problem seems to affect some users but not all a solution does not seem to be forthcoming, indeed it may not be an OpenTX issue at all but I do not what else to try,

schwabe, 2017-07-21:
This is probably a correlation that is not causation. The RX id and model id have a nothing to do with each other apart from the fact that the rxnum default is model id.

3djc, 2017-07-22:
I think @kilrah is the one who knows more about this, but I have indeed heard that some series of Taranis+ have issues binding high receiver number (doesn't seem to affect only SxR). It has indeed nothing to do with OpenTx , but hopefully he will give us more details

kilrah, 2017-07-22:
Anyone's guess what happens, but the RF module's unique ID is somehow mathematically combined to the receiver number to make the number that the receiver will store to know who it's bound to, and there's something that must be kinda broken in this process in module or receiver firmware.

Indeed not an OpenTX issue, and since we have no idea nor access to how all of this works you should talk directly to FrSky regarding that.

closed 2017-07-22 by kilrah

Alexander-C, 2017-07-22:
Cheers, I will contact FrSky.

Thanks again.