opentx/opentx #5104: S3 on Taranis X9D+ does not calibrate correctly & disappears from hardware config

CavRecon, 2017-07-22:
OpenTX 2.2.0
Reproduced on two radios: Taranis X9D+ & Taranis X9D+ Special Edition
Both radios have 6-position switch installed, configured as "Multipos Switch" for S3

While attempting to calibrate, the bar graph is very jumpy and inconsistent. [See example video here](https://youtu.be/t2NsXLB3wjE).

Sometimes, after usage or calibration, S3 disappears entirely from the hardware configuration. Suspect ADC scaling incorrectly.

Verified voltage across switch (Vcc - Gnd): 3.00v
Verified voltages present on the switch in various positions (Flap - Gnd):
1: 0.000v
2: 0.595v
3: 0.745v
4: 0.998v
5: 1.490v
6: 2.998v

kilrah, 2017-07-22:
Video looks as expected.
Disappearing during calib is possible if more than 6 positions are sensed due to a bouncy switch. No reason it would disappear under normal use though.

CavRecon, 2017-07-22:
Thanks for looking at this!

It must be the bouncy switch…although I rolled one of the radios back to 2.1.9 and it seems to be more tolerant. Despite being bouncy input, it seems to cleanly recognize the six separate positions on 2.1.9.

Did something change with S3 reading or calibration?

hrenz, 2017-07-23:
Same Problem with S3 at X9DPlus
6 Pos switch mounted 1 Year ago, works fine under openTx V2.1xx
got S31, S32, S33, S34, S35, S36
got -100, -60 -20 +20 +60 +100

But now under V2.2,0 final Verson nothing, it disapear in the hardware-setting
No S31 to S32
No  analog steps

Will see what happen at a X9E (I have one too)

hrenz, 2017-07-23:
Not one problem at calibrating the S3 / F3 with V2.19 at X9D+ and at X9E
calibrating works perfect
more tollerant, more Hysteresis, more time for detecting the 6Pos Switch at older openTx-Versions?

projectkk2glider, 2017-07-24:
I wonder why this is apparently working on the Horus and not on Taranis. I remember I did some refactoring around analog sampling and jitter filter and calibration code (this looks suspicious https://github.com/opentx/opentx/pull/3327/files#diff-38f78c082c0646df0f5cd9d302b7360dR83 :stuck_out_tongue: ).

CavRecon, 2017-07-24:
Honest suspicion: Perhaps the 6-position switches sold for taranis are lower quality and exhibit far more bounce than those on Horus? Perhaps older routine was more tolerant?

stanislavdavid, 2017-08-18:
I bought 6-position swtich from authorized reseller of frsky - rcstudio.cz and I do not think they have lower quality. It is the only service center for X9D+and I have the same issue

schwabe, 2017-08-18:
@stanislavdavid we are talking about orgina frsky ones, they are not well designed.

Stylusboy, 2017-08-21:
I've recently experienced the same 6-pos calibration failure. I installed the 6-pos switch option right after I acquired my Taranis a few years ago. Back then, OpenTx allowed effortless calibration of this switch. I've updated the firmware a few times (current firmware revision - 2.2.0 N359) and during these updates, I've never attempted to re-calibrate the sticks/pots/switches until last week.

I have numerous multi-rotor models setup on the Taranis and all use the 6-position flight mode switch. Following my attempt to re-calibrate the sticks/pots/switches last week, ALL of the setups ceased to work, due to the flight controller getting bad information (i.e. bad or wrong flight modes, etc.) and not arming.

I ran across this GitHub thread started by CavRecon and his experience was identical to mine. Kilrah's response got me thinking about the problem. Upon connecting the Taranis to 6-pos switch/resistor divider "wiper", I was able to easily see rapid and multiple switch bounces when moving to each of the flight modes. Although projectkk2glider's comment pointed to corresponding code, I wondered if the sampling rate or moving average filter rates had changed. I hung a small bit of capacitance between the "wiper" output and ground. The rapid switch bounce spikes smoothed out and without downgrading to earlier firmware revisions, the 6-pos switch then calibrated without issue and ALL of the tried and tested model setups returned to full functionality.  I will admit that I liked the earlier display of S3's position during calibration where an integer corresponding to the switch position was displayed under the S3 vertical bar. 

Surely a better quality multi-position switches could be sourced, but before the software/firmware changes, my 6-pos switch worked flawlessly. I decided to install a single SMT capacitor (size 0805) with a value of 0.1uF. This size of capacitor can be added either on the 6-pos switch carrier PCB or on the S3 input solder pads and the problem is immediately solved. Even if the sampling rates are returned to their earlier values, the capacitor won't affect such a change. I hope this helps.

Tom

[image]
[image]

Fraggs, 2017-12-05:
After updating my Taranis Plus to Opentx2.2 I also had the problem of the 6pos switch not calibrating (connected toS2) after lots of messing about I finally got it to calibrate 
While it isnt exactly a great solution by slowly (very slowly) turning/switching through all the positions I did like I said get it to calibrate, several times it missed the last position and the slider showed full one position to soon, but eventually I got it to calibrate correctly, I dont know if I was just lucky or its because its a newish switch thats not had much use.
I might look into the capacitor mod stylusboy mentions if no other solutions are going to be available

mpjf01, 2017-12-05:
It seems to work as before on my X9D. Is the value of the play delay relevant during the calibration process - or at all these days?

Pix4hawk, 2018-04-06:
same Problem with my X9e ..... thinking is clear ..... replace the switch..... nothing change..... 8-(
## **Thank you Tom !**
I use a "normal" size 0.1uF 

Mike

davidngrc, 2018-08-22:
I am on 2.2.2, can not calibrate, try many times already,
what is going one?
it was ok, why why why...

bsongis, 2019-09-10:
Cannot reproduce in 2.3

closed 2019-09-10 by bsongis

fiechr, 2020-01-03:
I've just installed a 6-pos switch and having the same issue: The switch disappears after calibration.

Will try adding a 100 nF capacitor between S and GND, like mentioned above.

[image]

fiechr, 2020-01-03:
Added a 100 nF capacitor at the other side of the PCB - calibration worked at first try. Thank you for the tip!

zedthirteen, 2020-11-24:
@Pix4hawk Capacitor modification on switch worked for me. Thanks Tom :-)