opentx/opentx #5105: Checklist Display Issues on Horus

xtrmtrk, 2017-07-22:
When displaying a checklist file on Horus (OpenTX 2.2) there is an extra blank line at the top of the display and text can only be displayed about 3/4 the width of the screen.  It would be nice to be able to use the entire display real estate.

Also, this is more of a feature request - it would be great to be able to display quote marks in checklists.  Short of that, could we get documentation on the font set available for checklist display.

3djc, 2017-07-23:
We have already discussed this, but I can't find the original thread. The issue is the font variable width : we cannot easily forecast the printed width, so we need to have a significant margin .

The font set depends on your radio language, but your standard Horus set is here : https://github.com/opentx/opentx/blob/2.2/radio/src/fonts/480x272/font_stdsize.png

Could you elaborate on quote marks (it is my english lacking here sorry) ?

[image]

xtrmtrk, 2017-07-23:
OK, now that I see how text is being displayed from an image, this all makes more sense.

It sounds like this is a settled discussion, but I'd still argue two things:  

First, the white space above the first line is a waste of valuable space.  The software should be able to easily detect the height of the font from the png image and adjust the position of the first line accordingly.  

Second, as computer users we're accustomed to text flowing off the edge of a page and we know how to deal with manual text wrapping.  I would argue that letting us use all of this valuable screen space effectively is more important than having the check-lists automatically movable across languages or platforms without needing to reformat the text.  I'd much rather be forced to do my own wrapping than loose this space.

The quote mark thing was my problem - it looks like my text editor was doing some weird simple ASCII to Unicode thing on quote marks.

Thank you for that link to the font png - I've been looking for something like that for a long time!  It'd be VERY useful to add this to the Lua manual.

I'm sure you know all of this - but here's some background on how I use checklists:  My checklists have 2 sections - a top section that displays all of the switch options and explanations (where I really want to display as much as possible), and a bottom section where I keep notes on all of the programming (which I don't even look at during flight time) I've done for the model.  I document things like the purpose of logical switches, notes on servo throws, the reasoning behind my mixes, etc.  I use 4 different types of flight controllers on about 20 different planes with different features on each and it's hard to keep track of what does what and remember why I set things up as I did without a "cheat sheet."  The checklist serves both of my purposes fairly conveniently.

xtrmtrk, 2017-07-23:
BTW, thank you for all the work you put into this.  I wish I had time to contribute as well.  Instead I donate money to the cause every year!

3djc, 2017-07-24:
Sorry, I have not been clear enough, trying again !
The issue as nothing to do with cross platform or cross language. The issue is that due to variable font width per character, we cannot say for example : 40 chars will fit exactly and get you to the other end of the screen. Some sentences with 45 will fit, some with 35 not (forget the exact numbers themselves, I take exemples). In this example, there is 10 of each letter:

[image]

Because we have no simple way to know where it will stop, we take a safe approach ensuring your sentences will be displayed.

I'll try to find some time to investigate about the first line

projectkk2glider, 2017-07-24:
What if we had a function that returns the pixel width of the given string?

Oldgibber, 2017-08-08:
While on this topic, a minor point, the checklist does not display on the Companion on startup, yet will do on the Horus screen. The checklist tick box is active, and the checklist can be seen with a long press of the menu button, so it is clearly there on the PC.

kilrah, 2017-08-08:
It should if you "simulate radio", but not if you "simulate model" (since that skips all startup stuff for convenience).

Oldgibber, 2017-08-09:
That's useful to know. Thanks.

closed 2020-04-06 by stale[bot]