opentx/opentx #5106: SD synchronization bag.

kinshas, 2017-07-23:
Windows 10, was checked 64bit and 32bit the same problem, SD synchronization bag. Companion and OpenTX 2.2! OpenTX and Comp 2.19 worked perfectly.
During synchronization the companion showed that message:

Starting synchronization: E:\RC Plane\SD TARANIS 2_2 -> D:\

Finished synchronizing E:\RC Plane\SD TARANIS 2_2 -> D:\ :
     New: 0; Updated: 0; Skipped: 1022; Errors: 0;
================================================================================
Starting synchronization: D:\ -> E:\RC Plane\SD TARANIS 2_2

Creating destination file: E:\RC Plane\SD TARANIS 2_2\System Volume Information\IndexerVolumeGuid
Copy failed: 'D:\System Volume Information\IndexerVolumeGuid' to 'E:\RC Plane\SD TARANIS 2_2\System Volume Information\IndexerVolumeGuid': Cannot create E:\RC Plane\SD TARANIS 2_2\System Volume Information\IndexerVolumeGuid for output
Creating destination file: E:\RC Plane\SD TARANIS 2_2\System Volume Information\WPSettings.dat
Copy failed: 'D:\System Volume Information\WPSettings.dat' to 'E:\RC Plane\SD TARANIS 2_2\System Volume Information\WPSettings.dat': Cannot create E:\RC Plane\SD TARANIS 2_2\System Volume Information\WPSettings.dat for output

Finished synchronizing D:\ -> E:\RC Plane\SD TARANIS 2_2 :
     New: 0; Updated: 0; Skipped: 1022; Errors: 2;

3djc, 2017-07-23:
https://answers.microsoft.com/en-us/insider/forum/insider_wintp-insider_files/cannot-delete-system-volume-information-folder-in/0b80793f-4d9a-4df2-808c-59a24835709e

Not a companion issue

kilrah, 2017-07-23:
The error is you having a "System Volume Information" on your SD card. You likely have formatted it NTFS instead of FAT32.

closed 2017-07-23 by kilrah

kinshas, 2017-07-23:
to kilrah SD formatted in FAT32!
to 3djc looks like you're right but still cannot fix!

3djc, 2017-07-23:
Yes, but 2.1 sync was way less sophisticated and less powerful. You have in your SD path things that should not be there, and this is what is causing the issue. Maybe there are a few things that we can do to help, but that will not cover any potential case

kinshas, 2017-07-23:
Hey, guys! My mistake! Just formatted SD and everything is fixed. Thanks!