opentx/opentx #5108: daylight savings time adjustment

geofrancis, 2017-07-24:
most countries in europe have a time adjustment twice a year, so in the summer here in the uk the clocks for forwards 1 hour so we are UTC+1 in the summer and UTC in the winter so all my logs usually end up 1 hour off.

https://en.wikipedia.org/wiki/British_Summer_Time

the period beginning at one o'clock, Greenwich mean time, in the morning of the last Sunday in March and ending at one o'clock, Greenwich mean time, in the morning of the last Sunday in October.

kilrah, 2017-07-24:
Summer time is handled differently in many countries, no reasonable way for us to handle all of that. 

Change the timezone when applicable.

closed 2017-07-24 by kilrah

projectkk2glider, 2017-07-24:
This really would be too complicated to implement. For example similar system on gnu/linux requires several updates per year just to keep up with the changes to the DST switching policies trough the world. https://launchpad.net/ubuntu/+source/tzdata/+changelog