opentx/opentx #5109: Wrong telemetry A2 voltage with FrSky D4R-II when fbvs-01 disconnected

michlalp, 2017-07-24:
A2 value is 10.8V which is correct, when I disconnect the fbvs it shows 9,6V but it should show 0. This behavior is only on D4R-II rx (2pcs tested). I tested it with D8R2+ and with this rx it works as expected.

kilrah, 2017-07-24:
When the receiver input is unconnected it is floating and will vary randomly. Nothing OpenTX can do about it. 

You probably want to leave the FBVS connected to the receiver, and disconnect the other side. Then you'll have 0.

closed 2017-07-24 by kilrah

michlalp, 2017-07-24:
it behaves the same if disconneted from rx or sensor, I'm 99% sure this worked correctly with 2.1

michlalp, 2017-07-24:
The value is 9,5V if sensor is disconnected from battery and connected to rx. Before turning the plane ON

kilrah, 2017-07-24:
Again there is nothing OpenTX-related in any of this. All we get is a number form 0 to 255 from the receiver, we convert it and display it. We have no way to magically guess the sensor is not connected.
As said when floating the value will be random. Sometimes it might be 0, sometimes something else.

michlalp, 2017-07-24:
Thanks for the info, I presume than only workaround will be modification of rx.