opentx/opentx #5112: [Horus] Reading internal GPS via LUA Script

labeh, 2017-07-24:
I like to read the internal GPS from the Horus via LUA Script to implement a Widget which can show me the distance between model (is also equipped with a GPS) and pilot. 

For this widget I need access to the internal GPS from the Horus via LUA. Currently I did much research including reading much opentx code but did not find any suggestion how the internal gps is available. Could anybody help me please?

schwabe, 2017-07-24:
Currently the internal GPS is not supported very well. You cannot it from lua or other telemetry. Search for ```INTERNAL_GPS``` and ```gpsData``` to see what is implemented so far. 

Making this available via lua should be easy but the question is rather if the internal GPS should available as normal telemetry source.

labeh, 2017-07-24:
I have done that already. But I sadly don't know enough about the rest of the code to do a quick fix and provide it by a pull request. It is not obvious for me how the data comes to the Lua Scripts. Maybe there should be a `LUA_EXPORT(...)` expression added in https://github.com/opentx/opentx/blob/cd23fedb56e85945455474d4e52d709740c0a114/radio/src/dataconstants.h ? But this can't be enough?

schwabe, 2017-07-24:
probably not.

As a quick hack, this might work, after the internal GPS position should available as telemetry value:

```diff
diff --git a/radio/src/gps.cpp b/radio/src/gps.cpp
index 5a62fb5ac..9b10694a5 100644
--- a/radio/src/gps.cpp
+++ b/radio/src/gps.cpp
@@ -265,6 +265,8 @@ bool gpsNewFrameNMEA(char c)
                 gpsData.altitude = gps_Msg.altitude;
                 __enable_irq();
               }
+              setTelemetryValue(TELEM_PROTO_FRSKY_SPORT, GPS_ID, 1, 0, gpsData.latitude, UNIT_GPS_LATITUDE, 0);
+              setTelemetryValue(TELEM_PROTO_FRSKY_SPORT, GPS_ID, 1, 0, gpsData.longitude, UNIT_GPS_LONGITUDE, 0);
               break;
             case FRAME_RMC:
               gpsData.speed = gps_Msg.speed;
```

3djc, 2017-07-25:
Yeah, GPS and lua need some love, I had a few things ready (like disatnce computation for lua), need to do some code archeology :D

projectkk2glider, 2017-07-27:
Should be easy:
 * export `MIXSRC_TX_GPS` to lua with `LUA_EXPORT()` macro
 * write custom handler in `luaGetValueAndPush()` for that source.

1Sven, 2017-07-27:
Perfect idea, the position and acceleration sensors are also essential. Then you can finally create a follow me mode.

WimSanders, 2018-03-07:
I would be already very happy if I can display the internal GPS.
So if the can make  as an selectable sensor in the telemetry menu it would be great.
If it’s only to find back a lost model

3djc, 2018-03-07:
All that is available

Was done in #5610

closed 2018-03-07 by 3djc

WimSanders, 2018-03-07:
Dear Open TX,

Thank you for replay.

I both my Horus 12s a half year ago, I’m very happy with it, but I still have to learn a lot.

I see that I have to do this by a Lua script, never used lua scripts before, but internet has all the answers. I gone try to find out how to do.

During this winter I have setup a sensor system based on Pawelsky libraries in RC-Groeps.

Now I have the GPS working. And want use the internal GPS as well.

With kind regards,

Wim Sanders

3djc, 2018-03-07:
You do not. You have been able to display internal sensor on widgets since 2.2.0 without any lua knowledge.

Create a sensor display widget and select the internal GPS, it is available as a source

WimSanders, 2018-03-07:
Thank you for replay.

I did but is say only Sats:0.

I waited for 30 minutes outside.

With kind regards.

Wim Sanders

3djc, 2018-03-07:
Either the radio could not get a fix (very cloudy, obstacles,...) or your internal gps is defective, because it does work

WimSanders, 2018-03-07:
Sorry did not wait not long enough, now I have a fix

THNX