opentx/opentx #5116: HORUS Not All DATA In Special-Function DELETED AFTER MODEL FLASHING OVER COMPANION

Githubbachelor, 2017-07-26:
Not All DATA In Special-Function DELETED AFTER MODEL FLASHING OVER COMPANION 2.2.0.

Today I want to store/save  all my models in Companion 2.2.0. They were stored under cocuments.

Than I load these models again back into my Horus which was OK.

BUT ALL MY DATA I prepared Special-Function were  deleted and not replaceable for example:

SF11 - VFAS -  Sag Wert ---- VFAS --- 30s
SF12 - ALT - Sag Wert ------- Alt -- 30S

All these data IN ALL MODELS are gone!!!!! I am very angry because I lost for 130 Models all data!

Other data like these are still there, for example:

SF5 - L03 --Say data --- Tmr1 -- 60s      or
SF9 - SA-  -- Say Text --- DuaRa1 (Dualrate1) --- 1x

What now???? I lost work of about 80h

Githubbachelor, 2017-07-26:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Githubbachelor, 2017-07-26:
These are the screens after flashing into Horus over companion. All functions with "SAY DATA" are lost. They were shown in the telemetry screen but lost in the special functions!

Githubbachelor, 2017-07-26:
Now I saw in companion that "ALL SAY DATA ENTRIES" are not transmitted to companion. That is the problem and I mean a VERY BIG BUG! And if you flash back the models over companion these data will be overwrite and and they are gone and never reproduceable.!

3djc, 2017-07-27:
You are very confusing, but I'm trying to understand. Are you meaning you lost the SF where a sensor was the switch ? Can you try a 'sag wert" SF with a regular switch ?

kilrah, 2017-07-27:
Can't understand what you did, what versions exactly you were using, and what the "source" of what you've sent to radio is

Githubbachelor, 2017-07-27:
Ok  I try to explain better:

You start sensor searching in Telemetry, and you get for example A2 and RSSI

When you prepare directly under SF:

 

SF1 – A2 – say data – A2 – 10s

SF2 – RSSI – say data – RSSI – 12s

 

That is working in your system HORUS Open TX official Version 2.2.0

 

Than you want to save your model with this entries over Companion 2.2.0 and you load your models into Companion. 

 

Than: Companion does take not these SF1 and SF2 – All Models lost there directly programed Data with SF A2 or RSSI or Curr and so on.

 

BUT:

Other data for example:

 

You prepare L1 a~x and L2 a~x  before and you set:

 

SF3 - L1 – say data – A2 – 10s

SF4 – L2 – say data – RSSI – 10s

 

Than companion does load these data from your Horus and you can see them in the opened Model in Companion.

 

And the problem is: You store all your models  with directly programed

 

SF1 – A2 – say data – A2 – 10s

SF2 – RSSI – say data – RSSI – 12s

 

In a file documents  and if you flash that document back into your Horus or other TX than all data were lost like

 

SF1 – A2 – say data – A2 – 10s

SF2 – RSSI – say data – RSSI – 12s

This.

 

The other data which were programed with L1, L2, L3, L4 aso… will be send to the Horus and work.

 

This Night I programed all my Models new with all SAY DATA in program mode with L1 – say data – A2 – 10s ---- for example

 

Maybe you can find a way that Companion take all programed data from Horus and did not filter the directly programed SF functions with direct A2 aso

 

Greetings Michael

kilrah, 2017-07-27:
Looks like using a sensor as switch has never been implemented in companion.

Githubbachelor, 2017-07-27:
It is possible to do that in the Horus and  you save much program work  and it is not complicated and easy and save time and most important memory in the system. Why so complicated to work in old style?  – IF Companion does take these data all is easier and you have more space for programing

kilrah, 2017-07-27:
It was forgotten. Seems you're the first to use this feature too since nobody noticed before.

Githubbachelor, 2017-07-27:
So you see that I am also interested in a good Software and Firmware and I like your work. I was also every time a pioneer is some products. What I am missing is a 

 

Android App for programing with Bluetooth. That is a Gold Joker

 

Now I prepared all last neght in old programing mode – now I am tired – veryyyy

 

I programed 138 models with complete telemetry

 

So hopefully you take that feature into the next 2.2.1

3djc, 2017-08-14:
Sensor switch implemented in https://github.com/opentx/opentx/commit/7d386e7821c2c7d7b382d81c6c54c7343db3a89b

closed 2017-08-14 by 3djc

Githubbachelor, 2017-08-14:
Thanks a lot !!!