opentx/opentx #5118: [Taranis X9D+] SD Card "read only": can't write, can't format

pboschi, 2017-07-27:
I have a strange problem with OpenTX 2.2.0 and 2.1.9 (not with OpenTX 2.0.6) on my recently bought Taranis X9D+: the microSD cards seems read-only when put in it!

1) I can't write to its partition when connected in bootloader mode (the Taranis freezes as soon as I try to - I have to remove the battery to turn it off);
2) I can't format the card from the SD menu page of the Taranis (it says "SD CARD ERROR")
3) ...but I can read correctly (sounds are working, I can navigate within the SD card page, I can read files from PC when connected in bootloader, etc - but I have to store those files using my PC card reader).

I've tried 3 different microSD cards (original, a no brand and a Sandisk), same result. They all work perfectly on my PC card reader.

Another user is reporting the same identical error:
http://openrcforums.com/forum/viewtopic.php?t=10285

3djc, 2017-07-27:
2. fix is already in and will be available in 2.2.1

pboschi, 2017-07-27:
Are you sure it is related? It's happening also on 1GB size cards, and i thought that fix was related to a bug on >4GB cards. But maybe I could get a more detailed error message with that patch. Too bad I don't have time to install and configure the toolchain and build it... I don't even have time to fly :) Do you have a Taranis build at hand of the current branch? I could check if it actually fixes or the specific error.

kilrah, 2017-07-27:
You might have a hardware issue with your SD card connector.

3djc, 2017-07-28:
I said "I can't format the card from the SD menu page of the Taranis (it says "SD CARD ERROR")" was fixed in 2.2.1, not the rest

pboschi, 2017-07-28:
The first thing I thought was a faulty SD card connector; but with OpenTX 2.0.6 it does work, and I've also noticed that logs and screenshot features are working, so the hardware is actually capable of writing (that's my naive assumption - I don't really know anything about low-level SD management).

So:
**2.1.9/2.2.0**
- Format: NOK
- EEPROM backup: NOK
- Bootloader-mode writes from PC / Companion: NOK
- Rename files: OK
- Log recording: OK
- Screenshot recording: OK

**2.0.6**
All OK

3djc, 2017-07-28:
Have you tried another HIGH QUALITY usb cable ?

pboschi, 2017-07-28:
I've tried different cables and even another PC, but there is no difference.
In fact, the fastest way to reproduce the problem is to try to save the EEPROM from the Taranis... which doesn't involve PC or cables.

I've also tried 2.1.6, after suggestion from FrSky support, with no luck.

At this point I only have 2 hypothesis:
1) it's an hardware problem... but a very subtle one, because with an older firmware all is working ok
2) the hardware has recently changed without notice and there is a compatibility issue with recent OpenTX firmwares

kilrah, 2017-07-28:
> it's an hardware problem... but a very subtle one, because with an older firmware all is working ok

It's possible current firmware checks the write protect line while older one doesn't.

pboschi, 2017-08-01:
After opening the Taranis, moving some wires, resetting the connectors, cleaning the SD card slot and recharging the battery, the problem disappeared. So indeed it was an hardware problem, or a poltergeist.

closed 2017-08-01 by pboschi

Paulvanmunster, 2017-10-15:
Hi wondering if I can comment on a closed issue. I am having exactly this issue with my new Taranis. Did your fix prove to be the solution. I am having no luck at all.

powercycle, 2017-10-15:
@pboschi, I find it interesting that you could still rename files, log and do screenshots from the radio.  Based on your comments it really does seem like a hardware issue, but there also seems like something else might be afoot.

pboschi, 2017-10-15:
I really don't know at this point, it is working perfectly since then. Maybe it was some kind of electrical fake contact... and renaming, screenshots and logs were ok because they involved very simple and fast operations on the FAT that didn't trigger the problem.

Paulvanmunster, 2017-10-15:
Hi guys , Just saw you were active this morning. this is really related to previous topic of pboschi where you could not write to sd of Taranis. I have just purchased a new Taranis and have exactly the problem you describe. Did resetting connectors and cleaning SD slot really turn out to b a permanent solution. I am having no joy. I have found that the only write function working with the card in the radio is to backup a model seems to work. All other write functions cause lockup and blank sd screen exactly as you described. Thanks.

Paulvanmunster, 2018-05-20:
Hi guys . I tore my hair out for ages and eventually replaced the sd card with a quality brand name card and all the problems went away. This was even after the radio was replaced under waranty and the replacement arrived with all the same issues.