opentx/opentx #5119: opentx 2.2.0 repurposing trim with extended limit

FlipAirboost, 2017-07-27:
Hi All,
Instead of increasing the adjustment range, the extended trim moves the neutral.
Example here with snapflaps.
First, we have more snapflaps as expected and then for the extended trim it becomes an offset.
[repurposing trim extended limit.zip](https://github.com/opentx/opentx/files/1181249/repurposing.trim.extended.limit.zip)

kilrah, 2017-07-27:
Please explain what radio type this is for, and what to do exactly with the switches, pots and stick to observe the problem. Not going to spend hours doing random things to try and understand what you're talking about.

FlipAirboost, 2017-07-28:
Ok , I'm trying
Companion 2.2, opentx 2.2.0, taranis x9d  Mike Shellim philosophy in mind.
[image]

First check trim extended
[image]
I use gas trim to adjust in flight snapflaps volume .
[image]
[image]
Volume trim minimum with full down elevator = zero snapflaps as expected.
[image]
Volume trim maximum with full down elevator = full snapflaps as expected
[image]
Volume trim maximum with centered elevator control = zero snapflaps as expected
[image]
Volume trim maximum + extended = more snapflaps as expected with full down elevator but also flaps with centered elevator control.
[image]

projectkk2glider, 2017-07-28:
Use multiply for the second lune of the CH11 mixer.

closed 2017-07-28 by projectkk2glider