opentx/opentx #5120: Companion Firmware Build Options Lua mouse over text update/change

RCdiy, 2017-07-29:
Current mouse over text "Support for Lua model scripts"
Suggest changing it to "Support for Lua MIXER scripts" 
This would clarify what that option does. 

https://www.rcgroups.com/forums/showpost.php?p=37914641&postcount=1102
The lua option is ONLY for mixer scripts. The lua engine is ALWAYS included and will run all other script types regardless of the option.

Maybe change "Lua" to "Lua mixer"

schwabe, 2017-07-30:
That is only true on the horus, on other platform it really lua

RCdiy, 2017-07-30:
The post was in a discussion involving the X7.

RCdiy, 2017-07-30:
From the Lua manual 

If you intend to use mixer scripts, when updating the firmware on your transmitter you need to make sure the lua option is checked in the settings for your radio profile (Main menu -> Settings ->Settings...) as shown below. This is not required if you only intend to run telemetry, one-time and function scripts, support for those is included by default.

RCdiy, 2017-07-30:
https://opentx.gitbooks.io/opentx-2-2-lua-reference-guide/content/getting_started.html

3djc, 2017-07-30:
And actually choosing that option gives access to "custom scripts" screen, and most people use it for many thing BUT mixer script, per say, so I'm not in favor of that change

RCdiy, 2017-07-30:
What change do we do then?
There is no "model scripts"
There are:
- Mix Scripts
- Telemetry Scripts
- One-Time Scripts
- Wizard Scripts
- Widget Scripts
- Theme Scripts
(Taken from https://opentx.gitbooks.io/opentx-2-2-lua-reference-guide/content/ Part 1 Script Type Overview)
No model scripts.

**How about
Current mouse over text "Support for Lua model scripts"
Suggest changing it to "Support for additional Lua features"
Maybe change "Lua" to "Additional Lua features"

In the manual we document:
"Selecting this options enables additional Lua features such as the CUSTOM SCRIPTS screen and mixer scripts. "**

RCdiy, 2017-07-30:
Apologies for all the repeated edits of the first post and title. Just trying to take feedback into account.

closed 2019-11-15 by stale[bot]