opentx/opentx #5123: Companion Crashes on startup

nveeser, 2017-07-30:
Companion 2.2.0 crashes on startup on OSX
Tried two different machines

Process:               companion [1210]
Path:                  /Users/USER/Desktop/OpenTX Companion 2.2.app/Contents/MacOS/companion
Identifier:            ???
Version:               ??? (???)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           companion [1210]
User ID:               502

Date/Time:             2017-07-30 10:28:24.721 -0700
OS Version:            Mac OS X 10.12.6 (16G29)
Report Version:        12
Anonymous UUID:        0F5BDB93-ECEB-495F-30B0-8BD759E544A7

Sleep/Wake UUID:       26A97155-9290-44C5-97C9-54F43619F7E6

Time Awake Since Boot: 7100 seconds
Time Since Wake:       2300 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       EXC_I386_GPFLT
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [0]

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   org.qt-project.QtCore         	0x00000001075c8109 QSettings::allKeys() const + 25
1                                 	0x0000000105c5b1ff AppData::importSettings(QSettings&) + 1695
2                                 	0x0000000105c57ba5 AppData::init() + 277
3                                 	0x0000000105985cf4 main + 420
4   libdyld.dylib                 	0x00007fff8c5ac235 start + 1

Thread 1:
0   libsystem_kernel.dylib        	0x00007fff8c6db44e __workq_kernreturn + 10
1   libsystem_pthread.dylib       	0x00007fff8c7c5621 _pthread_wqthread + 1426
2   libsystem_pthread.dylib       	0x00007fff8c7c507d start_wqthread + 13

Thread 2:
0   libsystem_kernel.dylib        	0x00007fff8c6db44e __workq_kernreturn + 10
1   libsystem_pthread.dylib       	0x00007fff8c7c548e _pthread_wqthread + 1023
2   libsystem_pthread.dylib       	0x00007fff8c7c507d start_wqthread + 13

Thread 3:
0   libsystem_kernel.dylib        	0x00007fff8c6db44e __workq_kernreturn + 10
1   libsystem_pthread.dylib       	0x00007fff8c7c5621 _pthread_wqthread + 1426
2   libsystem_pthread.dylib       	0x00007fff8c7c507d start_wqthread + 13

Thread 0 crashed with X86 Thread State (64-bit):
  rax: 0x5000000000000000  rbx: 0x00007fff5a359108  rcx: 0x0000000000000000  rdx: 0x00007fd2c2d02e48
  rdi: 0x00007fff5a359108  rsi: 0x00007fd2c2d02db0  rbp: 0x00007fff5a358c00  rsp: 0x00007fff5a358bf0
   r8: 0x0000000000000010   r9: 0x0000000105d04202  r10: 0x000000010adc4110  r11: 0x00000001075c80f0
  r12: 0x0000000000000000  r13: 0x0000000000000000  r14: 0x0000000000000000  r15: 0x0000000000000000
  rip: 0x00000001075c8109  rfl: 0x0000000000010246  cr2: 0x0000000105c5b15e
  
Logical CPU:     2
Error Code:      0x00000000
Trap Number:     13

Binary Images:
       0x1058a6000 -        0x1062f2fff + (??? - ???) <A7A92165-44D3-3F9C-8DE2-8A7229B674D5> /Users/USER/Desktop/OpenTX Companion 2.2.app/Contents/MacOS/companion
       0x106531000 -        0x106570ff7 +libSDL-1.2.0.dylib (0) <A8788186-446E-3B0D-9F62-7AFC93C26F17> /Users/USER/Desktop/OpenTX Companion 2.2.app/Contents/MacOS/libSDL-1.2.0.dylib
[... 270 lines cut ...]
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x13B), Broadcom BCM43xx 1.0 (7.21.171.130.1a1)
Bluetooth: Version 5.0.5f1, 3 services, 27 devices, 1 incoming serial ports
Network Service: Display Ethernet, Ethernet, en6
Network Service: Wi-Fi, AirPort, en1
PCI Card: Apple 57761-B0, Ethernet Controller, Thunderbolt@191,0,0
PCI Card: pci11c1,5901, IEEE 1394 Open HCI, Thunderbolt@190,0,0
PCI Card: pci12d8,400e, USB Open Host Controller, Thunderbolt@194,0,0
PCI Card: pci12d8,400e, USB Open Host Controller, Thunderbolt@194,0,1
PCI Card: pci12d8,400f, USB Enhanced Host Controller, Thunderbolt@194,0,2
Serial ATA Device: Samsung SSD 850 EVO 1TB, 1 TB
USB Device: USB 2.0 Bus
USB Device: Hub
USB Device: Keyboard Hub
USB Device: Gaming Mouse G400
USB Device: Apple Keyboard
USB Device: FaceTime HD Camera (Display)
USB Device: Apple Thunderbolt Display
USB Device: Display Audio
USB Device: USB 3.0 Bus
USB Device: MatekF4
USB Device: IR Receiver
USB Device: BRCM20702 Hub
USB Device: Bluetooth USB Host Controller
Thunderbolt Bus: Mac mini, Apple Inc., 26.1
Thunderbolt Device: Thunderbolt Display, Apple Inc., 3, 26.2

RCdiy, 2017-07-30:
Could you place the debug output in an attached text file? I once posted something similar and received a similar request.

I'm on macOS Sierra 10.12.6 (16G29) and have not had any issues.
Is there a forum thread somewhere on this? If not maybe post and I or someone else will try and help.
https://www.rcgroups.com/forums/showthread.php?2823315-OpenTx-2-2
or
http://openrcforums.com/forum/viewforum.php?f=45

RCdiy, 2017-07-30:
I just did a download from http://www.open-tx.org/2017/05/30/opentx-2.2.0 and have had no trouble running it.

nveeser, 2017-07-31:
I suspect that this is something to do with my preferences, or some such. Makes it hard to track down but still should not be happening

schwabe, 2017-08-01:
I think this bug is fixed in newer version. If you want, you can try this test build and see if it fixes your problem: http://opentx.blinkt.de:8080/~opentx/builds/opentx-companion-2.2.0test.dmg

nveeser, 2017-08-01:
Yes that seems to work for me. What was the issue?
FYI: Sierra 10.12.5 (16F73)

rhyde, 2017-08-12:
I am seeing a startup crash on OSX Yosmite.  I tried deleteing the previous preference files, but that did not change it.

schwabe, 2017-08-12:
@rhyde: your crash you posted in #5016 is also setting import related 2.2.1 should fix this (or the test version I posted here).

kilrah, 2017-11-08:
Duplicate of #5076

closed 2017-11-08 by kilrah