opentx/opentx #5128: SD Card Error Bootloader and Firmware

Grhosso, 2017-08-03:
Hello there,

I will explain my issues, maybe someone can help me.  I have Taranis X9D+ with 2.1.8 Bootloader and Firmware.

I followed all steps for install the OpenTX 2.2 and the SD contain but I get the Firmware 2.2.0 properly installed, although the bootloader is still on 2.1.8 (this is the first issue... I am confused between Bootloader and Firmware. When I trasnfer the firmware to Taranis, the bootloader does not change.

1. Do I have to install first the 2.2.0 bootloader and then the Firmware ??? I am confused. My current bootloader is 2.1.8. I have not seen a step by step about how upgrade bootloader and Firmware.

2. When I have 2.2.0 firmware properly uploaded from OpenTX companion 2.2, I always get a SD CARD ERROR but I have saved all SD content as required properly, all files.., v2.2  0010, I dowloaded the last version, and I continue getting this problem. why?

Thank you for your support.

3djc, 2017-08-03:
Sorry, this place is not one to get support, it is to track actual bugs. Please go to support forums, like https://www.rcgroups.com/forums/showthread.php?2823315-OpenTx-2-2 or http://openrcforums.com/forum/viewforum.php?f=45.

closed 2017-08-03 by 3djc

Grhosso, 2017-08-03:
Ups, really sorry. :/