opentx/opentx #5131: Firmware simulator final release 2.2 bugs

dadantsou, 2017-08-05:
Firmware simulator 2.2 final release bugs, so when I open Companion 2.2, edit my model then Simulate the model, **_servo outputs are not correct_**. It is a random bugs so **sometimes everything is ok, but if I restart Simulator the bug reappear**.

3djc, 2017-08-05:
Radio ? Os ? What should it ? What do you see as a result ? Whit the info you give, there is absolutely 0 chance that we can reproduce, and even less fix, the issue

dadantsou, 2017-08-05:
9XR PRO with opentx 2.2. Look for eg at this attached file, all my stick except throttle are centered but the outputs are differents.
[image]

dadantsou, 2017-08-08:
**FOUND THE ISSUE!**

My old Companion was not properly uninstalled so it creates conficts between OpenTX 2.2 and the older version. After correcting that, everything seems to be OK!

closed 2017-08-08 by dadantsou