opentx/opentx #5134: LS/RS Mixer/Source issue

PeterVRC, 2017-08-07:
Since changing to OpenTX2.2 I had never seen that in the Radio Settings you need to set the LS/RS to a value of "Slider". Mine were blank.
I had pre-existing models that use LS/RS as a Source in the Mixer lines. These continued to work fine after upgrading OTX2.1 to 2.2.
Today I loaded the models into OpenTX Companion to edit some things, and one thing was to CHANGE some values in a mixer line - this had LS as the Source.
I found that if I EDIT that entry when I exit out of it, the main Companion Mixer window loses all mixers from that edited one onwards! eg 8 mixer lines gone.
So there is some bug in there - which does not like you editing a mixer that has LS or RS if they were not actually enabled as Sliders int he Radio Settings.

Once I changed the Taranis to have the Sliders enabled, and reloaded the models into Companion, then it all edits as you would expect.

RipperDrone, 2017-10-03:
Similar problem here: When I load existing channel mapping including sliders, they show as an empty box in OpenTX 2.2. When I open the drop-down list, no sliders like LS, RS, ... are available for selection anymore.

Looks like the sliders are missing in the enum list to pick from. I hope that I have not messed up all my slider settings by saving from latest OpenTX back to the radio :-/

kilrah, 2017-10-03:
Needing to have the sliders defined correctly is necessary and intended, the issue seems to be that old settings are not imported correctly. As said above if you set them correctly you'll be fine.

RipperDrone, 2017-10-03:
How can I set them if the drop-down menu does not list any sliders at all? Confused... :-(

RipperDrone, 2017-10-03:
this is what the inputs page tab like (RS = Right Slider) still there...
[image]

RipperDrone, 2017-10-03:
Once I edit, it looks like this (empty source!): 
[image]

RipperDrone, 2017-10-03:
...and in the drop-down select box, there is no more RS, LS (sliders) available anymore :-(
[image]

RipperDrone, 2017-10-03:
Tried the same by setting up a NEW model - same behoaviour, no sliders shown for selection...

3djc, 2017-10-03:
There are missing in your RADIO settings

RipperDrone, 2017-10-03:
Ok - how can I fix this then?

3djc, 2017-10-03:
By going to your radio settings and ensure they are properly defined ?

RipperDrone, 2017-10-03:
You mean changing these settings to "slider with ratchet?"
[image]

RipperDrone, 2017-10-03:
I have not changed the radio profile whatsoever, seems it has been reset in OTX 2.2 latest version by default?

3djc, 2017-10-03:
Kilrah answered a while back. On a limited set of configs (we don't precisely which ones), those disapear during to move to 2.2.0. Just add them back and its done

RipperDrone, 2017-10-03:
ok, great. It worked - thanks a lot!

pghern, 2018-01-04:
I'm able to change the radio setting for the Sliders and then Store settings and save. but the next time I start Companion, the previous LS/RS setting are lost. I have to reset each time I open Companion.

Has this been fixed with the new 2.2.1 release??- I haven't updated yet, still on 2.2.0
Thanks,

kilrah, 2018-01-04:
> the next time I start Companion

What does that mean? With a blank file? After reloading form the radio or a file?

pghern, 2018-01-04:
After reloading Models and Settings from Radio

kilrah, 2018-01-04:
...and with what radio type?
Anyway update first.

pghern, 2018-01-04:
Taranis X9D+

Will update to see if that helps.
Thanks

mpaperno, 2018-06-02:
Was there ever a final determination on this?  Is there an import/conversion problem?

raphaelcoeffic, 2019-08-14:
@PeterVRC @RipperDrone could you please confirm or infirm this is still happening for you with the 2.3 nightlies (in case you have a supported radio)?

PeterVRC, 2019-08-14:
I am using OTX 2.2.3, but on a Horus X10S now, and I tried the above and it was all ok.
I will change to 2.2.4 soon. (I have not tried pre-release versions before.)
I can get out my Taranis and upgrade that and see what that does. In a day or two.

raphaelcoeffic, 2019-08-14:
@PeterVRC thx, trying out again with the new version (2.2.4) on your Taranis would be nice.

raphaelcoeffic, 2019-08-27:
> Was there ever a final determination on this? Is there an import/conversion problem?

Yes. We have unit tests for that now in 2.3 (incl. companion).

closed 2019-08-27 by raphaelcoeffic