opentx/opentx #5139: Cannot download new firmware for FrSky-X9D

Fojtik, 2017-08-08:
When I attempt to download a new firmware I obtain totally confusing message:
[image]

3djc, 2017-08-08:
Set your language in radio settings

Fojtik, 2017-08-08:
There is no oprion to set a language. I have everything in English.
http://open-txu.org/home/undergraduate-courses/radio-setup/beginning-radio-setup/

There is only option "Voice language" and the firmware does not allow me to change it.

3djc, 2017-08-08:
[image]

kilrah, 2017-08-08:
Set the correct radio type and it will be good again.

closed 2017-08-08 by kilrah

Fojtik, 2017-08-08:
Thank you for this. Anyway, before taking snapshot I already have English as actual language set.

After setting "Voice language" to English, it suddenly downloaded firmware correctly. 

But I suspect that there is something very unclean. The message box should at least write me which language it is guessing and which language should be set for proper functionality. Non english speakers might be confused by this anyway.