opentx/opentx #5144: Refactoring ComboBoxes in Companion

bsongis, 2017-08-09:
- have only one SwitchModel per model instead of one per tab
- only do the update work if something has changed

mpaperno, 2018-06-06:
Proposed new UI with collapsible grouping:

Sources:
[image]

Switches:
[image]

:+1:  / :-1:  :question:    

I had this working quite a while ago already, but... versioning got out of hand at the time. But I can port it.  

The tree view combo box is a [custom widget](https://github.com/mpaperno/maxLibQt/blob/master/src/widgets/TreeComboBox.h) I made.  It also uses a custom [proxy model](https://github.com/mpaperno/maxLibQt/blob/master/src/itemmodels/GroupedItemsProxyModel.h) which automatically groups items based on criteria (like switch/source type in this case), to create a tree from flat data.  So it all fits in perfectly with Qt item models.

mpaperno, 2018-06-06:
> have only one SwitchModel per model instead of one per tab

Definitely +1 on this, the question I always had though was who is to own it.  The logical choice would be the ModelData, but we're using memcpy (and such) to copy and move it around... which is not really going to work with pointers and child object ownership (and definitely not as a QObject with signals/slots/etc which would be very handy).  

So perhaps we need something like a "ModelManager" class which would be responsible for handling ModelData and all associated metadata for a model.  I think this would be easier (and probably better) than refactoring ModelData into a QObject.  Especially if one goal is to keep ModelData parallel/reusable with the radio version.

closed 2019-11-15 by stale[bot]