opentx/opentx #5145: Cels reads seconds from timer

Hotwire-, 2017-08-10:
Battery alert gets the timer for some reason and not battery voltage. 
See image:
[image]

tomkfly, 2017-08-10:
You Probbably have the wrong version of the SD Card.

kilrah, 2017-08-10:
Radio type? Other relevant details?

Hotwire-, 2017-08-10:
The gear is : Taranis X9D and X8R receiver and FrSky battery voltage telemetry module, OpenTx v2.2.0.

So the battery value (Cels) reports the timer, which is weird. This worked fine in OpenTx V2.0 and 2.1.

After deleting sensors and searching again the FrSky battery telemetry module, doesn't detect.  This was not due to any other changes, just the version of the firmware.

kilrah, 2017-08-11:
Cannot reproduce, your sensor probably has the wrong unit selected. When discovering the sensor will not show up until you connect a battery to it.

Hotwire-, 2017-08-12:
Apologies, the telemetry sensor had a broken track. No fault of V2.2

closed 2017-08-12 by Hotwire-