opentx/opentx #5146: R9M module Support for Taranis X9D/X9D+ user

seagovernor, 2017-08-11:
R9M module Support is just done in Ver 2.2.1.
Many Taranis X9D plus users prefer to use ver 2.1.x. Please, consider this matter.
Thanks a lot

3djc, 2017-08-12:
At this point in time, we are not planning to do any change anymore to 2.1.x unless a critical bug is found, which this is not

closed 2017-08-12 by 3djc

schwabe, 2017-08-12:
Also is the claim "Many X9D plus users prefer to use 2.1.x" somehow substantiated?

seagovernor, 2017-08-12:
Hm I understand you developers are doing lots of work without any rewards.
A Taranis user in my RC club just purchased R9M and R9 receiver set from Horus RC yesterday when it was in stock. So I told him it is not ready for Taranis X9D, but just for Horus with ver 2.2.x.
I believe many Tranis users assume it can be used, because FR Sky does not mention any special comment. 
Their Tx modules  so far have been used on Taranis without any doubt.  I am not complaining, just inform you. Otherwise I urge him to upgrade to ver 2.2.1 for R9M.

kilrah, 2017-08-12:
The R9M can be used with any radio, even non-FrSky, and as such with any software version as well.

You just don't get the convenient menu-based configuration and have to use PPM + the DIP switches that will be available in 2.2.1.

seagovernor, 2017-08-12:
Thanks for your clarification.
Thanks all the time developers.