opentx/opentx #5147: Deleted all models and reset itself to factory on Horus x12s

zjl715, 2017-08-12:
I am running opentxx 2.2.0 on a FRSky Horus x12s and today it randomly rest itself to factory. All models and setting were erased and the SD card still has jpegs I uploaded but the models folder has model one through 6 with no custom names. none of these models show up in the model selection. Does any one know how to recover or a reason why it did this?

3djc, 2017-08-12:
The radio and models data are stored only on the SDCard on Horus. It looks like there as been an issue with  your SDcard that damaged some of the critical files on your SDCard

kilrah, 2017-08-12:
> the models folder has model one through 6 with no custom names

This is normal, they never have custom filenames.
If you don't have a backup an index can be recreated to reference them.

3djc, 2017-08-12:
Closing this as this is not a bug

closed 2017-08-12 by 3djc