opentx/opentx #5154: Companion 2.2.0 Build Options Simulator ppmus

RCdiy, 2017-08-13:
The Outputs screen always showx -100 to 100 regardless of ppmus being selected... would be nice if in the simulator it changes with selection of ppmus...

Tested in X7 and X9D+ profiles.
[image]

RCdiy, 2017-08-13:
This image taken from taken from Tx

[image]

3djc, 2017-08-13:
The simulator is currently precompiled with mainstream options. We have discussed already a build on demand simulator, but we aren't ready for that

closed 2017-08-13 by 3djc

RCdiy, 2017-08-13:
Ah... I see...

Can't put an if statement in the simulator? Show depending on ppus selected? It seems to work for the noheli screen build option....

Just trying to understand...

kilrah, 2017-08-13:
The simulator is compiled once and for good with default options during the release process. It has no idea of what's selected in companion.

RCdiy, 2017-08-13:
@3djc @bsongis 
So why not add as to do instead of closing this?
I see an indirect reference in #987 but in a different context and the way the title is worded I doubt it will get much attention. Where as asking for a specific build option to be simulated may get someone interested in taking it on.

There are flight controllers where particular ranges of microseconds is needed so being able to simulate them would be helpful.

There is no need to compile a different similator. Could it not be a build options check and then either show outputs as percentage or as microseconds... A similar behaviour already exists with noheli where the simulator checks the build options before showing the heli screen or not...

3djc, 2017-08-14:
We want and maybe will do something like that at some point in time, to handle very critical things, like language for exemple. Once/if this is done, all options will be handled, but the complexity of this cannot be underestimated

sande005, 2018-02-11:
Another vote for this enhancement to have simulator follow profile build options, if ever possible some day....