opentx/opentx #5156: Option to boot directly to telemetry screen

papaxiclas, 2017-08-13:
Seems there is no way to boot directly to a telemetry screen?
If we are able to customize a telemetry screen for our needs I think it should be possible to power the radio directly to that screen instead of booting to a main screen that doesn't have any of the info we need. There are times that I find myself pressing the page button before even powering the radio.
Looking on rcgroups for this option a few users showed interest in this.
Thanks

PRGeno, 2017-08-14:
This would be a very useful feature.  I second the request.

bsongis, 2017-08-18:
We could perhaps do the same than what we have on X12S

papaxiclas, 2017-08-18:
And how is that?
I don't know the X12S:(

capitaineflam25, 2017-12-09:
It would be an improvement that I will appreciate too 👍

SoaringJim1951, 2017-12-12:
This is similar to my idea. I want a Special Function or a lua command to call a telemetry screen. e.g. On moving a switch a certain screen automatically comes up.

kilrah, 2017-12-12:
Didn't really know how to specify that, but I like the special function approach.

SoaringJim1951, 2017-12-12:
Thanks for your response. A lua script command that could perhaps be called
from a SF would also work well. Apologies for having no command of the
jargon!
Regards

<https://github.com/opentx/opentx/issues/5156#issuecomment-351115237>, or mute
the thread

notlebowski, 2018-12-07:
When using telemetry screens I noticed that during flying I often don't use them because I'm too busy flying. I do make use of Flightmodes a lot and together with certain FM's I would like a specific telemetry screen.
During FM1 (which is throttle lock) I'm interested in the Voltages of the pack and consumption, during flying I may be interested in height.

I would use it immediately in my models if it would be implemented in a way similar to this:

Only upon becoming TRUE the SF should switch to a certain screen or when the attribute is a GVAR and that changes.
Having a numeric attribute it could switch to a certain screen.
If a screen is not defined then do nothing...

Examples:

SF1 One Select Screen 1       # on model select go to screen 1
SF2 FM1 Select Screen 2      # in FM1 go to screen 2 (where you have consumption and voltages)
SF3 ON  Select Screen GV1  # When GV1 changes go to the appropriate screen if it exists
SF4 L01 Select Screen 0       # Go to home screen
SF5 L02 Select Screen +1    # Go to next screen
SF6 L03 Select Screen -1     # Go to previous screen

AndrewFarley, 2019-08-17:
Un-stale please

raphaelcoeffic, 2019-08-17:
Just a proposal: wouldn't it be good enough to remember the screen (main or telemetry page number) that was displayed last for each model?

AndrewFarley, 2019-08-17:
> Just a proposal: wouldn't it be good enough to remember the screen (main or telemetry page number) that was displayed last for each model and return to that screen whenever the model is loaded (either on boot or when the model is selected)?

Absolutely

bsongis, 2019-08-17:
it makes sense, I think the telemetry screens could be considered as main views

pompecukor, 2019-08-19:
would be nice to add this feature please.

SoaringJim1951, 2019-08-19:
I am still interested in this enhancement. Most useful when practising for
F5J .
Jim Hathaway

3djc, 2019-08-19:
We had a long discussion, and our proposal is the following:

- On main screen, you will navigate through 'general view' followed by telemetry screen (with back/forward using PAGE/PAGE LONG)

- Current PAGE LONG jump to telem screen disapears

- On model setup screen : you will have the choice of wich screen is displayed at boot (any of the 4 general + 4 telem), and also AUTO to remember your last screen displayed

AndrewFarley, 2019-08-19:
@3djc This would be amazing and would suit my needs perfectly.  In glancing at various of the other comments/desires above I think it would cover everyone's pretty well.  If you start work, feel free to ping me I'lll happily test.  I have some custom telemetry dashboards I always start up on model choice and/or on handset boot (which, is the same as model choice technically just for the last used model).

If you need a tester let me know, drop a line here I'm sure some of us will gladly help.

bsongis, 2019-08-19:
I prefer "LAST" over "AUTO"

fabioeh, 2019-08-23:
Would also like to see this feature implemented.

kilrah, 2019-08-24:
It is.

pompecukor, 2019-08-24:
> 
> 
> It is.

where bro?

3djc, 2019-08-24:
Radio work is done (but not merged yet), companion work still to be done. It should be in 2.3 when it releases

lshems, 2019-09-21:
When will this be in a RC? Would love to test.

3djc, 2019-09-22:
We have moved that change to 2.3.1

Jackjan4, 2019-10-03:
Really looking forward to this!

SoaringJim1951, 2020-01-19:
I looked at 2.3.5 but can't see this feature?

3djc, 2020-01-19:
As you can see, it has not been merged, and won’t be in 2.3. It should be in 2.4, in one way or another. A few things still need to be defined

Alwinator, 2020-09-25:
I am also waiting for this feature! (2.3.9) I hope it will be there soon!

git2323, 2021-09-21:
I am also interested in this. Thanks.

bigtwin, 2021-09-21:
Would be nice to have it already in 2.3 :-)

mcuprojects, 2023-03-30:
Any chance to see it in 2.4?

lukegluke, 2023-09-07:
Would also like to see this feature implemented anyhow, at least as simple parameter. No need in radical changes.