opentx/opentx #5160: Taranis radio - Re-Order pick list when adding telemetry sensors

mpjf01, 2017-08-14:
When adding a telemetry sensor value to a Taranis screen using the radio the user is presented with a large number of items in a list to choose from and it is necessary to scroll through them to near the end to get to the sensor data items. It would speed things up for most if the sensor data item could appear first in the list. If that's not possible or desirable for some reason then some means of getting to them more quickly would help. This option may be dealt with in #4713  Re-ordering is preferred by me.

schwabe, 2017-08-14:
You can hold enter long to jump to telemetry sensors.

mpjf01, 2017-08-15:
Thanks, I looked at that, Enter long jumps to another menu which allows you to choose the telemetry sensors as the last entry. When I am trying to populate sensors onto a telemetry screen on the radio it's the sensors I want to see first. I'm trying to suggest that it would be helpful if there was a way that avoids too much scrolling and unnecessary keystrokes. The process now seems to employ a standard mechanism and the ordering does make sense for some selections, it's just not ideal for the telemetry sensors. Would it not be feasible to treat the telemetry sensors as a special case, activated when the user has one of the telemetry entry screens open?

flarssen, 2017-08-15:
#2540 would help a little.

kilrah, 2017-08-15:
> Enter long jumps to another menu which allows you to choose the telemetry sensors as the last entry.

Which means you can get there with a single press of the UP key.

closed 2017-08-15 by kilrah