opentx/opentx #5162: Crossfire LUA for Horus

IdefixRC, 2017-08-16:
Hi guys, 
Opening another request as the previous one was closed (later not realized as initially planned). 

Looking for support to generate a working Crossfire LUA for the Horus. 

Happy to support with testing and basic coding (limited LUA know how). 
Hope to find someone who has know how and interest to realize it with me. 

Thanks

SoulPictures, 2017-08-22:
@IdefixRC, yesterday I played around a bit with the script, sadly I don't have hardware at hand... but I got it mostly running (as far as I can test), could you test these two files?

[CF-modded.zip](https://github.com/opentx/opentx/files/1241167/CF-modded.zip)

IdefixRC, 2017-08-22:
Great. Thanks. I will test shortly and let you know. 
I also did some work on it but I'm stuck at the device page view (parameter taranis page view vs horus screen implementation). Let's see if we get this to work :-)

IdefixRC, 2017-08-22:
@SoulPictures, I have tested your code. 
The changes you made to the crossfire.lua work. As you do not have a CF yet you will not have been able to try the device.lua. The main issue with it is that it uses code which is not available in the Horus implementation. 

Main issues I believe are as follows:
- Uses getLastPos() which is not available in Horus (your device.lua crashes because of that)
- Builds the parameter list and displays it based on paged with 7 lines each. This does not work for the Horus somehow (does not display at all). I have therefore looked at using the full Horus display size with 22 parameter lines (11 lines in 2 columns). While this works it crashed due to some loop issue I have (I think).

I have forked the code and created a branch and uploaded my current working code.
- crossfire.lua difference compared to yours: Included a header, color differentiation and a blinking message in case no Crossfire is found.
- device.lua includes the same header (with the device name) and my dirty hacks above (look for --Note to see what I changed.

Link: https://github.com/IdefixRC/opentx/tree/Horus-Crossfire/radio/sdcard/horus/CROSSFIRE

Hope you have some time to take a look. Would be awesome if we get this to work together (looks like the Horus user base is not the biggest from the response rate here :-) )

Some pictures:
[image]
[image]
[image]

IdefixRC, 2017-08-22:
And the error message after it builds the 2nd column
[image]

SoulPictures, 2017-08-25:
Got my Hardware today, will test around with a friend but not sure how far we will get

SoulPictures, 2017-08-25:
Trying out your script gives me sometimes bot TX&RX but often just one of them, the elements stay "..." after selecting one of them. Is this the Version from your Screenshots you linked there? Sometimes I get "CPU Limit" as an error message

IdefixRC, 2017-08-26:
I noticed the same although most of the time i get both. Hope the horus does not have the same issue as the x7 :-(

The cpu error comes from the loop error in the device script i think. The same reason why it crashes after displaying the parameters. 
The version on githib which i linked above is the same i made the Screenshots with. Had no time to make more changes in the meantime :-(

Cuxx, 2017-08-27:
I think the two coloumns mod is not enough. Channel Map is to long. you would need atleast 3 columns. but this would not fit.

IdefixRC, 2017-08-28:
true, could be an issue as we get max 22 lines with it.
Have to figure out how to implement multiple pages on the horus :-( (among other things...)

Cuxx, 2017-08-28:
Nope. I just skipped the first 10 variables to reach the output1 & output2. So i was able to change the Microv2 to CRSF-protocol. At the moment I don't have the time to figure it out :(

stevenoa, 2017-08-28:
Great work, I lack the knowledge to hack these scripts but I do have a taranis 9D, Horus, Crossfire Original TX, the new crossfire mini (JR module) so I will be willing to run anything you guys have available and provide feedback if you need it.

IdefixRC, 2017-08-31:
Gents, please test the updated script here with your Crossfires and let me know if it works for you (worked for my MicroTX): https://github.com/IdefixRC/opentx/tree/Horus-Crossfire/radio/sdcard/horus/CROSSFIRE

Note: I believe there is some speed issue with the Horus, maybe similar to the issue with the QX7 as the parameters sometimes take a very long time to populate. So after you access the device, wait a bit while the parameters load and populate the screen. If you scroll down once the screen is full (11 lines) it should scroll the screen to reach the next parameters.

Screenshots of the device page from the current release:
[image]
[image]

IdefixRC, 2017-08-31:
ran some more tests including the MicroRX, Diversity RX, configurations, binding and wireless firmware update. All are working fine :-) Look like this is good for release 👍

IdefixRC, 2017-08-31:
#5190

SoulPictures, 2017-08-31:
Kind of works here, parameters get shown, but I did not see parameters scroll. I'll test tomorrow if I can scroll with a unit with more parameters. At least with my TX unit (the bigger one) I didn't had any scroll function visible, simple bottom of the list was top of the list as next element

stevenoa, 2017-09-01:
Here are my results:
Crossfire Mini TX - everything working from what I can see with both a micro RX and a diversity RX.
Crossfire Original large format TX - detects the TX, select the TX screen sits blank for a while never seems to get a list of anything, just blank. If I bind an RX with the rear controls, it will then detect the RX in my case it was a Diversity RX, select the RX and it sits there blank again. With the large format TX I can use the menu on the TX so the LUA is not critical, having full functionality for the Mini TX is great progress, now I can setup the mini TX from my Horus. Obviously it would be nice to have the large format TX working as it does work on the taranis without problems.

stevenoa, 2017-09-01:
@IdefixRC for the protocol details differences etc Perna at TBS would be your point of contact from what I understand Crossfire is his baby and he is looking for somebody to help as he does not have a Horus. As I said getting the Mini TX working is a big win, the original Crossfire TX would be the icing on the cake :-)

sniperxnl, 2017-09-09:
I tested it also today and i get most of the time CPU LIMIT problem

IdefixRC, 2017-09-10:
@sniperxnl, can you specify whether you tested with  the microtx or the large one?

I'm in touch with TBS on the issue with the large TX and currently the issue lools like it points to a similar problem as the QX7. Will keep you guys posted.

sniperxnl, 2017-09-10:
it is with the xjt version with latest firmware

sniperxnl, 2017-09-18:
I hope we get fast a stable version, myn crashed every time

IdefixRC, 2017-09-18:
Have not yet received feedback from TBS. Have you tried with the latest tbs fw? It has a fix for the qx7 which might also help here.

Fromnboobto, 2017-09-23:
I have tried the latest verison that was linked in here, still not getting it to work at all:( I was not expecting this at all tbh, Wich is sad as f.... atm, my crossfire.lua shows "f-Micro RX(i had bound it before attempting lua)", I "choose" the F-Micro RX and it crashes out to "?:0, Attempt to call field "getlastpos"(a nil value.... (cant see more)
Device.lua Shows a blank screen for ages and nothing happens..

Any suggestions?

IdefixRC, 2017-09-24:
Guys,

Please provide more info.
- Which crossfire TX module and which FW version are you running on it.
- Which Horus OpenTX FW version are you running?
- What crossfire devices are you trying it with (RX, OSD...)

For the lua please use the published one in the opentx main repository.

Known limitations: not working with the large crossfire TX module. In
addition it can happen that the script does not find the TX module (RX
modules show up only). Investigation on these issues is ongoing. Currently
it is believed to be a similar Frsky HW/crossfire SW issue as with the QX7
(just not as severe hence the random issues)

airsupplyaerial, 2017-09-28:
Hey Guys,

Any Progress on this? I'm here to help test anything. (Rather inexpereinced) Horus X12s Open TX 2.2.0 and TBS Crossfire Standard TX and Diversity RX V1.57. Running a Pixhawk 2.1 brain. Able to communicate via Mavlink PPM, but nothing on Horus...

IdefixRC, 2017-09-30:
Waiting for feedback from TBS on the horus issue in order to continue testing (idea is to test the same fix that was used for the qx7)

stevenoa, 2017-09-30:
Details of FW version etc as requested.
OpentX - 2.2.0 2017-05-30 (I did look for a newer version a bit ago but could not find one).
Modules - Mini and full-size modules.
TBS version - 2.06 (did not go to 2.07 as it has reduced the bandwidth according to feedback)
Problem - mini module works with the Horus X12, fullsize does not, but as it has the oled on the back I can configure it that way so its not a show stopper.

airsupplyaerial, 2017-10-01:
Ok. Thanks for the reply. I'll await some hopeful
News!

Wyatt

noahwaldner, 2017-10-14:
I have the big Crossfire, why is it not running? Maybe i can help.

airsupplyaerial, 2017-10-14:
That’d be awesome.

Barbaryann, 2017-10-31:
Hello,
I have the Horus X10S and the big Old Crossfire, The Lua script not working.
https://www.youtube.com/watch?v=1GC48kuu1yI&index=4&list=PLmUct0I1--liC4f3-cGAO_j6Lwbq-y0Ai

closed 2017-12-18 by kilrah