opentx/opentx #5165: Crossfire LUA for QX7

justinmader, 2017-08-17:
Howdy,

Would love to see a working LUA for the Taranis QX7. The one for the X9 doesn't work properly (freezes immediately). Happy to help test!

Cheers,
Justin

IdefixRC, 2017-08-18:
Hi Justin,
I'm currently playing around with the scripts for Horus and QX7.
When you say it freezes, do you mean it shows "waiting for crossfire devices" and does not show the TX or connected RX? or does it crash for you?

justinmader, 2017-08-18:
Sometimes I get the "waiting for devices" message. If tx, rx, or core do show up in the list, selecting either will make the screen go blank and stay that way until I exit out.

techsavvyfpv, 2017-08-18:
For me it doesn't freeze.  I just can't scroll down to select anything besides failsafe.  I think it's because the X7 uses a scroll wheel instead of buttons.

justinmader, 2017-08-18:
That's what happened when I first tried it, would freeze on Failsafe and scroll wouldn't work. Now I can't even get that far. Have been trying different combos of FW on the Core Pro and Micro TX, all to no avail. But yeah, a friend says the LUA should work when it's adjusted for the screen size/scroll wheel of the QX7.

techsavvyfpv, 2017-08-18:
[CROSSFIRE.zip](https://github.com/opentx/opentx/files/1235176/CROSSFIRE.zip)

Here's what I got so far.  You can try it and see if it let's you scroll

IdefixRC, 2017-08-19:
Interesting. On my X7 it sits at searching for crossfire for a few minutes, after that the devices slowly show up but that's it. Thought it had something to do with the telemetry issue mentioned somewhere else.

OhmEye, 2017-08-20:
FIW, the script is not able to reliably talk to the TX due to #4922 but once that is resolved it looks to me like the script still needs work to support the X7 rotary input and smaller screen. I have done the inverter mod to my X7 and the script does run and appear to talk to the TX just fine, but the UI can't navigate the menu or change options, which makes sense to me given the different button input the X7 has.

techsavvyfpv, 2017-08-20:
@OhmEye Since I haven't done the mod yet could you try the crossfire.zip I posted above and see if it works for you.

OhmEye, 2017-08-20:
@techsavvyfpv actually I just saw this issue when I posted and missed that you had a link, sorry! I just finished my own mod of the script and my version works fine for me now. I just needed a working X7 to test on.

FWIW, my effort is at  http://ohmeye.com/download/debug/X7_CROSSFIRE.zip

techsavvyfpv, 2017-08-20:
I just tested it on my X7 and it works.  Awesome job OhmEye!!

OhmEye, 2017-08-20:
@techsavvyfpv FWIW I just took a look at your script and it doesn't take into account the smaller screen on the X7 so is drawing the right column entirely off the screen (which can't be seen.) I had to change the x coordinate for everything in that column to fit it all on the screen properly. I didn't load it on my X7 to actually test since mine is working fine.

techsavvyfpv, 2017-08-20:
Yeah my script wasn't working.  Your's works for me.  Thanks alot.  I can finally use my Crossfire Micro TX with my X7.

justinmader, 2017-08-21:
Cheers @OhmEye works for me too. Nice one!

OhmEye, 2017-08-21:
After feedback/testing and a minor update yesterday I think the X7 version of the script is fully functional. Pull request #5176

justinmader, 2017-08-22:
I tried it again last night and couldn't get it to work at all unfortunately. Got the "Waiting for Crossfire" message, powercycling over and over.

OhmEye, 2017-08-22:
@justinmader and you have working CRSF telemetry on an early version X7 or one with the mod? The most likely cause for the script timing out is the CRSF hardware issue.

justinmader, 2017-08-22:
@OhmEye I've never tried to get the audible telemetry warnings as I use the OSD. I do have CRSF protocol selected in Opentx and it _did_ work for me the first day I tried it, just not yesterday. Nothing has changed on my radio between now and then.

OhmEye, 2017-08-22:
@justinmader The X7 issue doesn't affect CRSF control, just telemetry. If you do not get constant "Telemetry Lost" warnings and can see a solid RSSI on the telemetry screen when you are using CRSF, then your unit isn't showing the symptoms. The nature of the issue may make the symptoms intermittent on some units.

I don't know enough to guess if a firmware update will be a desirable fix, if the solution is to drop the CRSF data rate by half. I support making it an option for people who would rather trade some latency to avoid modding the hardware, but I personally prefer to do the mod, which gives me full speed CRSF and has me working *now*. Fortunately, it seems there is now a simpler mod possible with a single resistor. If I were doing a mod today I would try that method first.

I'm no expert but with the information I've read at this point, there can be no firmware fix for the hardware issue. There can only be a hardware fix, or a firmware workaround that will involve significantly reducing the speed that CRSF communicates with the devices. In my opinion, the inverter mod is a hardware fix, and the simpler resistor mod is a hardware workaround, but both result in full speed CRSF with no changes on the TBS end. Any firmware mod will require changes on both ends and have reduced performance compared to X9D and other remotes using CRSF.

mtnmikev, 2017-08-28:
So does the QX7 need the Mod to have a working Crossfire LUA? I have tried two different lua with no luck.

OhmEye, 2017-08-28:
@mtnmikev The X7 does need to have fully working CRSF for the script to talk to the module. A common symptom of the CRSF issue is "telemetry lost" warnings. If your unit is affected by the issue (it seems most are, all the ones with the s.port present next to the USB port) then you will need to either do the mod or wait for the firmware fix which may be coming very soon.

mtnmikev, 2017-08-28:
Ahhhh! Thanks. So yeah I will do the mod.

gustavlindgren0, 2017-09-15:
Thanks for your efforts but I'm stuck at "Waiting for Crossfir..." then it disappear out of the screen windows

OhmEye, 2017-09-15:
@gustavlindgren0 You are aware that CRSF communication with the module does not work on a stock X7 or with the current 2.2.0 release of OpenTX and have read all the comments above? Did you do a hardware mod to fix CRSF on your X7, or are you running the firmware update with the baudrate selection? If so, which baudrate do you have selected?

Hinecken, 2017-09-30:
I have done a successful hardware mod to my x7 and am currently using this script.  I writing to double check if everything is working as it should compared to other Crossfire / X7 users.   I am able to communicate with both the module and the rx.  It takes a few seconds to load, but works 90% of the time.  Occasionally selecting the Rx will cause the screen to blank.  Most of the time it works and my options all slowly load.   The quad flies and I have full telemetry.  The only warning I get is "SENSOR LOST" when I am disarmed.  It does not warn me immediately, appears to be at random actually, and I have only heard it while disarmed.

spuder, 2017-12-25:
[This script](https://github.com/opentx/opentx/pull/5176) works with 2.11, but I can't get it working with 2.12. The screen just goes blank

[image]
[image]

seanthesheep101, 2018-07-03:
Hey @OhmEye I’ve updated the Qx7 to 2.2.1 and used your crossfire lua script because the original one that came with the sd card contents wasn’t working (screen wouldn’t load when selecting MicroTx) so when I used your one the same thing happened where the screen stayed blank but sometimes it would slowly load one or two options and I could scroll between them but couldn’t do anything :( how do I get this working?

kilrah, 2019-01-06:
Continue in #6287

closed 2019-01-06 by kilrah

spartanz51, 2019-01-15:
Fixed it with software modification only:

1) Flash Latest OpenTX Firmware
2) Open Menu, go into Hardware page
3) Set Baudrate to 115200

janiofk, 2020-01-03:
Hello OhmEye

I have just bought new Tbs crossfire micro tx and i have a taranis QX7, i've updated my firmware for 2.3.4 in taranis and crossfire update last version.

i cannot get this working, can someone help me ?

select tx micro  then just stop working, once i scroll the roll button on the right to aside give me this script error.
Script killed
CPU limit.

does anyone having the same ?

3djc, 2020-01-03:
What is your Crsf module version ?

janiofk, 2020-01-03:
> What is your Crsf module version ?

XF Micro TX 3.23
HARD: V 1.03

3djc, 2020-01-03:
Could you replace the file in /SCRIPTS/TOOLS/CROSSFIRE by this one and test again ?

[device.zip](https://github.com/opentx/opentx/files/4020506/device.zip)

janiofk, 2020-01-03:
> Could you replace the file in /SCRIPTS/TOOLS/CROSSFIRE by this one and test again ?
> 
> [device.zip](https://github.com/opentx/opentx/files/4020506/device.zip)

script killed
device.lua:101
cpu limit

O.o

3djc, 2020-01-03:
Have you set your telemetry speed to 115k ? RADIO/HARDWARE/Max Baud set at 115200 (nearly at the bottom)

janiofk, 2020-01-03:
magic, thanks so much 3djc, much appreciated.

what i did was to replace the one you send to me + use an old crossfire.lua that i download on the top and the baud worked.

could be my memory card reading problem.

thanks mate!

3djc, 2020-01-03:
Have a lot of XFIRE fun !

hutch81, 2020-08-27:
I'm stuck on waiting for devices with the qx7. I did the hardware mod ... Nothing talks.  It won’t find crsf in lua or crsf setting.I can’t get stick inputs to show up in BF config on Rx tab. Running 2.2.4

3djc, 2020-08-27:
If you don't get proper telemetry, the script won't work, but this sounds like you have a hardware issue

hutch81, 2020-08-27:
Hardware on the module or the radio? And what could be causing the telemetry issue....mod  done incorrect?

3djc, 2020-08-27:
The mod incorrrect can obviously be an issue, but e do not have cristal balls, really hard to say without physical access. If you have the module with Oled, yo ucan check that at least the module does get pulses in RC Input line

hutch81, 2020-08-27:
Well I put the module into my x9 and it works so I'm going to assume it's either my mod or telemetry like you said but I followed the mod directions exactly. So what may be causing the telemetry issue? Thanks in advance

3djc, 2020-08-27:
the mod

hutch81, 2020-08-27:
Right so if the baud rate is set at 115200 then I must hav botched the mod somehow

hutch81, 2020-08-27:
So now I see that some newer qx7 didn't need the mod. How do I know if I did the mod and didn't need to and also what do I do if I did the mod and didn't have to

githubnw, 2020-10-04:
I carefully mod'd my QX7 following the GetFPV YouTube instructions for my newly acquired XF Micro TX V2 module. Thinking/hoping going with TBS Crossfire instead of Frsky's R9 solution would save a lot of headache. However, after much effort my QX7 just keeps getting stuck at the "Waiting for Crossfire" message when I executed **crossfire.lua** from the **SCRIPTS\TOOLS\CROSSFIRE** folder. The message is not displayed at the center of the QX7's LCD screen, so the last 2 characters of the "Crossfire" word are cut off at the right edge of the LCD.  I am pretty sure I updated to the correct Frsky model's (QX7) firmware version 2.3.9 (**opentx-x7-multimodule-noheli-lua-en-2.3.9-otx.bin** - hoping to get the latest LUA scripts), along with the latest SDCard file contents (**sdcard-128x64-2.3V0029.zip**). Searched all over the Internet but found very little information - most pointed to the mod being needed. BTW, I can already bind to the XF Nano Rx (using the TBS Agent X Windows app) but want to be able to configure the XF Micro Tx V2 using the QX7. Being totally new to the Crossfire world I was hoping some experienced users could offer some advice or trouble-shooting steps, before I suspect having received a flawed Tx module and have to go thru the trouble of filing a RMA with TBS on this new purchase. Any help would be much appreciated...thanks.

3djc, 2020-10-04:
Starting with the mod was a bad bad idea, because now it is impossible to know if your CRSF module is ok or not. If your internal module is set to off and external set to CRSF, then it is very likely your mod. It could be the module too, but it is less likely

githubnw, 2020-10-04:
3djc, thanks for your quick response. I forgot to mention that my QX7 was stuck at the "Waiting for Crossfire" message even before I did the mod. I did the mod carefully and used the inverter provided in the TBS Crossfire Micro V2 Starter Kit. I've built many quads in the past (all with Frsky RX though) and did carefully check the wiring/soldering of the mod to be correct. Per all YouTube and user forum info I've gone through so far I cannot figure out what I might have missed or done wrong.

3djc, 2020-10-04:
Before you did the mod, did you set MAx Baud to 115k ?

githubnw, 2020-10-04:
3djc, no, I changed the baud to 115k only after I did the mod and after reading the suggestion above in this thread. I was naive enough to expect simple deployment of this XF product, and the TBS quick-start guide didn't mention about the baud rate change. However, I believe before I made the change the default was at a Max of 400,000(?), and I thought that should cover the lower 115k baud rate mentioned in this thread.

3djc, 2020-10-04:
No, without mod, the radio needs to be set to 115k (CRSF  works very fine at that speed too, unless you are into high end racing, I doubt you can notice any difference). With the mod, it can be set at 400k

jerambooh, 2020-10-05:
I have a simular issue. I can read my crossfire.lua and chance something in micro TX, but my screen gets stuck at nano RX and I get a message 'script killed cpu limit'. I have a qx7 and did the mod carefully. If i press on nano RX I only see ... ... ... Any suggestions would be great.

hutch81, 2020-10-05:
I had the same issue....I had the mod inverted. Flipped it over resoldered
wires and it worked perfectly. Check and make sure you didn't do the same
thing.