opentx/opentx #5168: Make bluetooth wireless trainer compatible with ersky9x

MikeBland, 2017-08-17:
Using HC-05 or HC-06 bluetooth modules I've had wireless trainer working over bluetooth for some time in ersky9x. I use a SBUS frame with 16 channels.
I feel it would make sense if openTx used the same format and number of channels so bluetooth wireless trainer would work between them.

Mike.

kilrah, 2017-08-17:
The BT trainer we're implementing at the moment is meant to be compatible with FrSky OS's implementation on the Horus. 
Would rather suggest ersky9x changes instead for compatibility to be as wide as possible ;)

bsongis, 2017-08-18:
I have one of those modules, but it is not BT-LE, am I right in saying they won't be able to communicate either with Horus or even with Taranis X9E?

projectkk2glider, 2017-08-18:
@bsongis I think you are right. Remember we even can't get X9E and X12 to communicate, even when they both have the same BT LE modules (but obviously different firmware).

MikeBland, 2017-08-18:
I can't test with production firmware on the Horus as mine is a prototype (V1.0 I think).
I'll need to run some tests with the X9E as well (swapping the battery from the Horus!).
The HC-05/HC-06 are recognised by my PC with BLE bluetooth. I understood that BLE was supposed to be back compatible.
Can anyone e-mail me a datasheet with the commands for the BT module now shipping on the Horus? My datasheet only has the TTM commands.
I'll consider adding this BT trainer format, but I don't want to downgrade from 16 channels to 8 channels!

bsongis, 2017-08-18:
If I am not mistaken BLE is not at all compatible with standard BT
It should be possible to add a configurable channels count, provided the maximum length of a BLE frame is not reached!

kilrah, 2017-08-18:
Note that Horus prototype (depending on what prototype actually) should still be able to run current firmware, just need the "pcbdev" option selected on OpenTX (PCBREV=10 in cmake). FrSky OS should autodetect hardware rev.

MikeBland, 2017-08-20:
I've done some testing. X9E with openTx the BT is recognised by my PC as a BLE device, so no serial and would need an application written to use it.
Horus prototype with openTx 2.2 (pcbdev option), no BT device recognised by my PC. I have enabled it, but maybe it is in master mode and I haven't (yet) seen a way to change it to slave mode if necessary.
Ant datasheet for the BT module in the Horus available?

projectkk2glider, 2017-08-21:
>Ant datasheet for the BT module in the Horus available?

Nothing official from FrSky (I believe), we have basically reverse-engineered the commands and trainer protocol by listening on the BT module - CPU interface in FrOS.

kilrah, 2017-08-22:
Horus BT code isn't in 2.2.0 yet, you have to build current 2.2. branch to have it.

closed 2017-11-08 by kilrah