opentx/opentx #5170: make: *** No rule to make target 'opentx-taranisplus-simulator'.  Stop.

adelton, 2017-08-19:
Hello,

following the instructions on https://github.com/opentx/opentx/wiki/OpenTX-2.2-Linux-Build-Instructions in Fedora 26 container, the compilation fails with

```
[100%] Built target companion22
make: *** No rule to make target 'opentx-taranisplus-simulator'.  Stop.
```

The full log is

```
[root@3167e0e582c6 /]# cd opentx
[root@3167e0e582c6 opentx]# mkdir build-taranis-debug && cd $_
[root@3167e0e582c6 build-taranis-debug]# cmake -DPCB=X9D+ -DGVARS=YES -DLUA=YES -DDEBUG=YES -DCMAKE_BUILD_TYPE=Debug ../
-- The C compiler identification is GNU 7.1.1
-- The CXX compiler identification is GNU 7.1.1
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Deprecation Warning at CMakeLists.txt:14 (cmake_policy):
  The OLD behavior for policy CMP0023 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

-- Found Git: /usr/bin/git (found version "2.13.5") 
-- OpenTX 2.2.0 @42cbc568
-- Qt Version: 5.7.1
-- Found FOX: /usr/include/fox-1.6  
-- Foxlib found at /usr/lib64/libFOX-1.6.so
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found SDL: /usr/lib64/libSDL.so;-lpthread (found version "1.2.15") 
-- SDL Lib: /usr/lib64/libSDL.so;-lpthread; Headers: /usr/include/SDL; Version: 1.2.15
-- Found PythonInterp: /usr/bin/python3.6 (found version "3.6.1") 
-- Python found, version: 3.6.1
TARGET simu/libsimulator: cpp compiler /usr/bin/c++ v7
CMake Warning at radio/src/tests/CMakeLists.txt:46 (message):
  WARNING: gtests target will not be available (check that GTEST_INCDIR,
  GTEST_SRCDIR, and Qt5Widgets are configured).

-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/cc
TARGET bootloader: cpp compiler arm-none-eabi-g++ v7.1.0
TARGET firmware: cpp compiler arm-none-eabi-g++ v7.1.0
-- Simulators library search path: /usr/local/lib/companion22
-- QT_TRANSLATIONS_DIR not found, unable to package pre-built Qt translations with application.
-- Qt lupdate not found, 'translations' target will not be availabe.
TARGET companion22: cpp compiler /usr/bin/c++ v7
CMake Deprecation Warning at companion/src/CMakeLists.txt:327 (cmake_policy):
[... 379 lines cut ...]
[ 93%] Building CXX object companion/src/CMakeFiles/companion22.dir/moc_creditsdialog.cpp.o
[ 93%] Building CXX object companion/src/CMakeFiles/companion22.dir/moc_releasenotesdialog.cpp.o
[ 93%] Building CXX object companion/src/CMakeFiles/companion22.dir/moc_releasenotesfirmwaredialog.cpp.o
[ 96%] Building CXX object companion/src/CMakeFiles/companion22.dir/moc_customizesplashdialog.cpp.o
[ 96%] Building CXX object companion/src/CMakeFiles/companion22.dir/moc_splashlibrarydialog.cpp.o
[ 96%] Building CXX object companion/src/CMakeFiles/companion22.dir/moc_splashlabel.cpp.o
[ 96%] Building CXX object companion/src/CMakeFiles/companion22.dir/moc_progresswidget.cpp.o
[ 96%] Building CXX object companion/src/CMakeFiles/companion22.dir/moc_progressdialog.cpp.o
[ 96%] Building CXX object companion/src/CMakeFiles/companion22.dir/moc_process_copy.cpp.o
[ 96%] Building CXX object companion/src/CMakeFiles/companion22.dir/moc_process_flash.cpp.o
[ 96%] Building CXX object companion/src/CMakeFiles/companion22.dir/moc_process_sync.cpp.o
[ 96%] Building CXX object companion/src/CMakeFiles/companion22.dir/moc_flashfirmwaredialog.cpp.o
[ 96%] Building CXX object companion/src/CMakeFiles/companion22.dir/moc_flasheepromdialog.cpp.o
[ 96%] Building CXX object companion/src/CMakeFiles/companion22.dir/moc_downloaddialog.cpp.o
[100%] Building CXX object companion/src/CMakeFiles/companion22.dir/moc_mdichild.cpp.o
[100%] Building CXX object companion/src/CMakeFiles/companion22.dir/moc_mainwindow.cpp.o
[100%] Building CXX object companion/src/CMakeFiles/companion22.dir/moc_radionotfound.cpp.o
[100%] Building CXX object companion/src/CMakeFiles/companion22.dir/moc_wizarddialog.cpp.o
[100%] Building CXX object companion/src/CMakeFiles/companion22.dir/moc_modelprinter.cpp.o
[100%] Building CXX object companion/src/CMakeFiles/companion22.dir/moc_multimodelprinter.cpp.o
[100%] Building CXX object companion/src/CMakeFiles/companion22.dir/moc_modelslist.cpp.o
[100%] Linking CXX executable ../../companion22
[100%] Built target companion22
make: *** No rule to make target 'opentx-taranisplus-simulator'.  Stop.
```

3djc, 2017-08-19:
Instead of "make companion22 opentx-taranisplus-simulator", do:

make libsimulator && make companion22

adelton, 2017-08-19:
Thanks, https://github.com/opentx/opentx/wiki/OpenTX-2.2-Linux-Build-Instructions updated.

closed 2017-08-19 by adelton