opentx/opentx #5171: Strange behavior of telemetry display with Multimodule

pboschi, 2017-08-19:
Hello, maybe I'm doing something wrong, but I've noticed a strange behavior on the display of telemetry values of Hubsan X4 104C and Blade Nano QX using the IRangeX Multiprocol module (latest 1.6.21 fw) on Taranis X9D+ (official 2.2 fw). I think it's more related to OpenTX, so I open the issue here.

**Hubsan X4 10C**
- A1 is correctly populated in Telemetry page (with correct battery voltage)
- A1/A1+/A1- are correctly populated as numbers when selected on Display page
- A1 is **not** correctly read on LUA scripts using getValue("A1")
- A1 is **not** displayed on top bar, even if such option is selected on Display page
- RSSI is not working - but that's maybe due to the model not sending it, no problem

**Blade Nano QX**
- RSSI is displayed correctly on top bar
- RSSI is **not** listed on the Telemetry page (there is maybe a somewhat related TRSS value, but i can't relate it to RSSI)
- RSSI is **not** correctly read on LUA scripts using getValue("RSSI")
- RSSI can **not** be referenced in any way on Display page

schwabe, 2017-08-20:
DSMX has not a real RSSI value. So the best we have is the rssi of the received telemetry (TRSS) and that should be used on the Spektrum models. 

For the Hubsan the A1 should really work like expected. 

When you say not correctly displayed/not correctly read, what you do exactly mean by that? Are you getting a different value or none at all?

pboschi, 2017-08-20:
Thank you for your answer. 
Ok for the DSMX. 
For the Hubsan, on the main screen I simply don't get the top bar displaying of A1... I only get the Taranis battery display. On the scripts, 0 is always returned. If I create a bars/nums telemetry page with the A1/A1+/A- values included, they are displayed correctly.

schwabe, 2017-08-20:
Iirc the multimodule with Hubsan just outputs Frsky d hub telemetry. There is no special handling of this case. So it is really strange that it does not work for you. Have you tried renaming the sensor?