opentx/opentx #5172: 32 Telemetry fields is not sufficient.

Devil-67, 2017-08-20:
Hello,
I'm new to Github and hope I'm right here.
First of all thank you very much to the programmers of OpenTX. Big performance !!!

I use the OXS sensor, RPM / Temp, FLVSS and the Redundancy Box.
My problem is that the 32 telemetry fields are not enough.
Is it a possibility to expand to about 40 - 45?

Greeting
Mario

powercycle, 2017-08-20:
Wow.  From the looks of it, Redundancy box has output power for each servo.

Devil-67, 2017-08-20:
Here is a composite screen shot from the required fields of Companion.

[image]

kilrah, 2017-08-21:
You still have 2 free it seems. And do you really use all of them? E.g usually not useful to extract all cells, jsut the minimum and the difference.

geofrancis, 2017-08-22:
i have reached and breached the max sensor limit when using telemetry converters along side some frsky sensors, the most annoying thing is that the handset becomes unusable when it happenes, since it throws up the error about too many sensors every time another sensors over the 32 is discovered, so if i have another converter active it wont let me exit the screen till i press exit 20+ times to clear each extra sensor.

Devil-67, 2017-08-22:
At the moment I am the first, I think there will be more.
But that's the future, there was no such thing in the past, but today everything is possible. ;-)

More and more sensors are coming onto the market,
Then the Diy sensors like OpenX Sensor and the whole Arduino Diy
Which offers very many measuring possibilities and very many users build it after and like it.

The whole is of course also very helpful in the setup of drives.
Also the flight security RSSI, lost frames, voltage, current, altitude, distance etc.
Then the position, in the case of an outside landing, I fly only big thermic gliders in the mountains.
Then so helpful things like Compensated Vario (thanks Mstrens) Airspeed and much more.
If I do everything here enumerate, it is perhaps a bit much.

kilrah, 2017-08-22:
We'll see. Problem is this would use significant amounts of RAM, which could cause trouble with popular Lua scripts not being usable anymore on X9D. And yet more EEPROM usage of course. So if implemented X9D might have to be left out.

> the most annoying thing is that the handset becomes unusable when it happenes

Power receiver off, do your cleanup, power on again.

Devil-67, 2017-08-22:
If there are really problems with the Lua, the user can decide whether the script or the telemetry values is more important.
Even a compromise with a little less than 50 fields would be a big win.
If of course 50 fields work, then please the 50 fields ;-))
Perhaps there is also a possibility that the user
A selection can make how many fields needed
And then selected and transmitted in the companion.
Thank you for your participation in my problem

schwabe, 2017-08-22:
letting the user decide is diffcult. On an embedded platform like OpenTX you try to avoid dynamic memory allocation. A sensor needs at the moment 45 bytes (13 for eeprom+ 32 current values)  of ram. 

That does not sound much but we have only 128 kB Ram on the Taranis of which 66 kB already used without lua even running. 32 extra sensors would take an extra almost 2kB of RAM.

Devil-67, 2017-08-22:
There need not be an additional 32 fields, 18 are sufficient.
But I'm also happy and possible with any other field you can provide.
How to do it best I would like to leave the specalist,
The variables assignable was just an idea.

Gewe99, 2017-08-22:
I have a big glider with 6KW jet  impeller..
Actually  I am using 31 telemetry values..
I have some self developed sensors...
And GPS is still missing...
I also would like to have more values . I am using a Horus12s.
64 Values would be great... 48 could be a compromise.
Perhaps there should be a compiler option if RAM space is an Issue on smaller radios...

Please give us more...
Thanks and regards Dirk

schwabe, 2017-08-22:
compiler options are problematic as you end up with incompatible eeproms which is a nightmare.

RealTadango, 2017-08-23:
Please expand to 256 values for the Horus / X10 :) It 256 is to much i can accept 128 also ;)

Devil-67, 2018-07-06:
Hello everybody,

2.3 is not far away and I would like to revisit the topic.
I think it will now also several users give this extension want / wish

OpenUAS, 2018-08-16:
If you really really want it, the best possible option is to make the code enhancement, test it yourself until you are satisfied, and make a pull request. Then the discussion likely will have more success.

Devil-67, 2018-08-17:
yes, I really want that.
But unfortunately I am not able to change the code myself.
I have to leave that to the specialists who are very knowledgeable about it.

3djc, 2019-05-02:
We are really struggling for memory on Taranis, yet, we do acknwoledge there is more and more telemetry comming in.

Our current plan is to push in 2.3 the sensor limit to 40 for Taranis, and 60 for Horus. Horus should also get a boost in maximum channels given the increase ACCESS does provide, but that is something we cannot afford on Taranis

raphaelcoeffic, 2019-08-14:
already implemented in 2.3.

closed 2019-08-14 by raphaelcoeffic