opentx/opentx #5174: Special Functions - Parameters - Source - Inputs

beNative, 2017-08-20:
When a model's inputs are edited, changes are not reflected in the sources dropdown list on the tab 'Special Functions'.
When you for example add a special function to adjust a GV, and select 'Source' in the Parameters section, the dropdown list does not show the changes.
I'm using version 2.2.0 of OpenTX Companion.

rgds,
Tim

powercycle, 2017-08-21:
My testing shows sources come direct from hardware or the mixes screen.  Inputs don't seem to be available in the special functions screen in that scenario.

I was able to assign an input to a mix, then that appeared as you were expecting.

3djc, 2017-08-21:
@beNative please could you provide us a step by step scenario we could follow tu understand better ?

Thanks

kilrah, 2017-08-21:
Try closing and reopening the model editor. It's known a lot of stuff isn't updated live at this point, there's already an issue open for it.