opentx/opentx #5188: Create new cpp function that can be used by lua script

jujuducoin, 2017-08-29:
Hello I'm new in openTX (I've not received my horus yet, bought today)
I would like to know the clean way to create my own cpp functions in order to use it in lua scripts.
So which file(s) I have to modify ?
Maybe later I will try to make an mp3 player
Thank you

projectkk2glider, 2017-09-01:
@jujuducoin why did you close this? Are you still interested? I can write a small howto.

jujuducoin, 2017-09-01:
I closed because it is not really an issue ^^
But if you can do this it would be really nice !

projectkk2glider, 2017-09-01:
First I would encourage you to discuss your ideas for new functions before actually implementing them. This will give you a broader view and much more chance of you final code being included in the official OpenTX. You can talk to us on https://opentx.rocket.chat/channel/OpenTX_General

Now to the implementation:

Each Lua API function is:
 * declared in appropriate table (each module has one such table) - this provides a link between Lua function name and C function name
 * implemented as actual C function that does something.

Lets take your example `crossfireTelemetryPop()'. The function belongs to the general part of OpenTX API and is defined in https://github.com/opentx/opentx/blob/5b5c88fb4ec96ffa2dbb66fd8ae76febb895ab0a/radio/src/lua/api_general.cpp#L1201 

As you can see the C function is named `luaCrossfireTelemetryPop` and is implemented in the same file at https://github.com/opentx/opentx/blob/5b5c88fb4ec96ffa2dbb66fd8ae76febb895ab0a/radio/src/lua/api_general.cpp#L411

How the function itself is implemented:
 * function gets parameters that were passed to it from Lua function call (this exact function is a bad example since it doesn't take any parameters, but look a other functions). 
 * it then calls appropriate existing OpenTX functions 
 * prepares result (a return value of Lua function)

You can read more at https://www.lua.org/pil/26.1.html

jujuducoin, 2017-09-01:
For the moment I'm just trying but after that why not to purpose some changes
Thx you save my time :)

closed 2019-11-15 by stale[bot]