opentx/opentx #5196: Fewer sensors using Copter 3.5.2 than 3.4.6

pmshopUSMC, 2017-09-05:
After upgrading to Copter 3.5.1, fewer sensors show up on Taranis X9D + with Open TX 2.2 loaded.
Example: Copter 3.4.6 produced RSSI, RxBt, VSpd, GPS, Alt, and VFAS.
Copter 3.5.1 produces RSSI, RxBt and GPS only.
Same for 3.5.2

3djc, 2017-09-05:
Did you change opentx version as well? Because it sounds like a Copter issue, not opentx

schwabe, 2017-09-05:
I have no idea what Copter is but the problem seem to be Copter related since its version change sees to have triggered the problem.

pmshopUSMC, 2017-09-05:
@schwabe - Ardupilot Copter firmware 3.5.2 and yes, 3.5.x does seem to have triggered it. That is why i am reporting it for development.

@3djc opentx version 2.2 is loaded

schwabe, 2017-09-05:
@pmshopUSMC You realise that is the OpenTX issue tracker and not the Ardupilot bug tracker?

pmshopUSMC, 2017-09-05:
Yes, I do realize this.
If Ardupilot made a change, doesn't Open TX need to follow suit to keep the previous operations, correct?
I mean Craft and Theory does the same with their Flight Deck .lua

3djc, 2017-09-05:
No, it is their task to send the sensors in the right format

kilrah, 2017-09-05:
Most likely you now simply need to manually enable what you want to send in your arducopter configuration...

closed 2017-09-05 by kilrah

pmshopUSMC, 2017-09-05:
So it appears that the mavlink protocol was tweaked a bit and appears open tx hasn't caught up yet...so my guess Ardupilot wont comment...it will be up to Open TX to adjust

schwabe, 2017-09-05:
@pmshopUSMC we don't even support mavlink telemetry, so telemetry is probably in sport format.

pmshopUSMC, 2017-09-05:
Then what delivers the sensor data in OpenTX2.2?

schwabe, 2017-09-05:
if you use frsky receivers as I said sport (smart.port). But this is not a support forum.

pmshopUSMC, 2017-09-05:
Right...not a support forum...I am reporting to the OpenTX team that the telemetry now sent by copter 3.5.x does not trigger the "Discover Sensors" as it used to. Used to pick up 6 sensors and now only picks up 3

schwabe, 2017-09-05:
And I am telling you that smart port format is unchanged and if OpenTX does not discover the sensors anymore something is wrong what Copter is sending as sport.