opentx/opentx #5201: Horus V2.20 Trim-Steps allways fix at 4us   at X9D,  X9E  I can set it and works 1, 2, 4, 8us

hrenz, 2017-09-09:
Hy,
after testing setting trim steps at Horus  I was surpriced
Trimsteps are allways only 4us  independend of setting at modell

At X9D, X9E  I can set trim steps ultrafine, fine, mid, .... and so on  
and I get Steps wth 1us, 2us, 4us, 8us  and expo thats ok since years

I have tested with an Impuls with measering unit

Please chek it yourself

[image]

3djc, 2017-09-09:
You are doing something wrong. The 4 settings exists on Horus, and they are workling as expected,

3djc, 2017-09-09:
1 step coarse (so 8)

[image]

[image]

[image]

3djc, 2017-09-09:
1 step fine (so 2) 

[image]

[image]

[image]

My center is at 1.501ms

hrenz, 2017-09-10:
ok I see,
have tested again with Horus and X9D+ 

This 4us steps are only at throttle  trim and only when "Trim idle only" is set.
At the other trims I can set it 1, 2 ,4, 8 as expected

Yesterday at my tests I had tested only the throttle  (ch1)
at Horus the trim idle only was set random
at X9D+ the trim idel only was not set random

so my error, thanks for testing

closed 2017-09-10 by 3djc