opentx/opentx #5204: ImmersionRC EzUHF SBUS no longer works after upgrade from 2.1.6 to 2.2.0

rdww60, 2017-09-11:
It seems the channel ranges for DMS2/DMSX have been fixed at CH1-6 in 2.2.0, and was CH1-8 in 2.1.6. Taranis Plus, with IRC EzUHF JR TX module. This has affected both my RaceFlight Revolt V2 configurations and BetaFlight BrainFPV RE1 configurations. FrSKY L9R 12 channel SBUS receiver using internal RF works fine. Just EzUHF broken.

schwabe, 2017-09-11:
You are leaving out important details like what Transmitter (X9D?) and how you configured the external module in OpenTX

If you are using DSM then the code has not really changed between 2.1.6 and 2.2.0 and has always been 6 channels.

kilrah, 2017-09-11:
Doesn't make any sense, don't see how DSM would have anything to do with EzUHF.
DSM has always been 6 channels (companion had a bug and allowed setting 8 at some point, but that never actually had any impact).

schwabe, 2017-09-11:
Quick google search shows that EzUHF allows to use the DSM2 protocol as serial protocol. If you want more channels via Serial protocol, EzUHF should implement one of the other serial protocols we have implemented (PXX, Multi or Crossfire). E.g. ORLS is using Multi as protocol (https://github.com/opentx/opentx/issues/2894).

Closing this issue because DSM2 hasn't changed.

closed 2017-09-11 by schwabe

rdww60, 2017-09-12:
Please don’t be so short, I am your customer, and trying not only to get my Taranis working under 2.2.0, but also to help you guys improve your software.

I did not provide too much information because it is a lot of effort, and often, it is a waste of my time as there is usually no response.
Thank you for responding.

I have reverted to OpenTX2.1.6 through backups using companion. All my hardware is working again, and I see the CH1-8 selection for the DSM2/DSMX selection in the external RF setup for the model (I could only see CH1-6 in OpenTX2.2.0 as confirmed by your remarks).

I would like to provide you with information to replicate my problem, and I’m wondering what could be useful to you?

1)	Radio FW and Settings backup .bin for 2.1.6 (included FW and Radio backup (“pre”)
2)	Radio FW and Settings backup .bin for 2.2.0 (included FW and Radio backup “post”)
3)	Build Information:
•	Transmitter: Taranis X9D+
•	Transmitter Module: EzUHF JR (with SBUS firmware)
•	Receiver: EzUHF Nano or 4CH Lite (with SBUS firmware, non-inverted)
•	FC: BrainFPV RE1, TX3, 5V, Inverted, SBUS (with BetaFlight 3.2)
•	FC: Raceflight Revolt V2, TX1, 5V, Inverted, SBUS (with RF1 317 or 325 .bin)

I can provide pictures of just FC and receiver connected to configurators if relevant.

Summary: When I upgrade from OpenTX2.1.6 (stock from factory), to OpenTX2.2.0 (download), ALL my EzUHF SBUS hardware becomes non-functional (uses JR module). ALL my FrSKY hardware is still functional (uses Taranis internal radio). And, as I noted in the bug/issue submission, the only observation I have from a layman’s point of view, is the DSM channel range selection changed from CH1-8 to CH1-6. It appears to me that this is an issue associated with IRC EzUHF configuration that worked in 2.1.6 and doesn’t in 2.2.0, and the issue is repeatable in that I can go back and forth between OpenTX versions.

I’m surprised this hasn’t already come up, but I do acknowledge that finding someone who as actually got the IRC EzUHF SBUS implementation to work is rare, the instructions aren’t very clear. Hopefully you have relationship with IRC to discuss the issue in more detail.

Any other information that would be helpful to you? Please confirm if you want  pictures too.

Regards,
Brian Page

[Radio_backups.zip](https://github.com/opentx/opentx/files/1297354/Radio_backups.zip)

schwabe, 2017-09-12:
Yes, I know that 6ch vs 8ch. I actually fixed that bug. The CH1-8 vs CH1-6 was purely cosmetic. The code never outputs and never has outputted more than 6 channels. The only other change that happened between 2.1 and 2.2 was that I fixed the output of DSM to be 100k 8e2 instead 100k 8e3. (Just an extra unnecessary pause between two bytes). And the code still works for DSM2 hack modules as we have feedback that these still work (e.g. every DSM2 module that is used in Horus). So we are pretty sure that the DSM2/DSMX protocol is still working. 

We don't use IRC but you are welcome to join us on http://opentx.rocket.chat

rdww60, 2017-09-12:
Hi Arne,

Thank you for the reply. IRC = ImmersionRC company that makes the EzUHF products, VTXs, Vortex Quads. Sorry. Do you have any relationship with them to discuss if this change could be what causes things to break? 

Thanks in advance.

 

And I’d like to join, I’m sure to learn something. Thanks for the invite.

schwabe, 2017-09-12:
No, I have no relationship with them. And I don't think any of the other OpenTX developers has any EzUHF Module, so we have absolutely no way to test this. You should probably contact IRC and see if they are willing to work on that or they are willing to contact us to work something out so we can fix this.

Btw. their page also mentions only 6 channels (https://www.immersionrc.com/ezuhf-firmware-v1-53-setup/)

rdww60, 2017-09-12:
Yes, what they don’t tell you is that you have to still transmit 8 RF channels, that was the trick, transmit 8 but utilize 6. That’s what it took to get it working. I have a large commitment in their products, not just their receivers, so I think I will initiate discussion with them at FPVlab.com, which is their preferred support site. Would you mind if I reference this dialog to possible spur a reaction on their part?

 

Thanks again.

schwabe, 2017-09-12:
Sure reference this dialog.

rdww60, 2017-09-13:
I don’t suppose it would be possible to get a software knob to change that DSM pause back? I have no idea how much trouble that would be, but would be a quick way to see if that is the issue. I really think is it, since the SBUS via FrSKY L9R is working fine in both, it is 12 channels.

 

I also invited/posted at FPVlab, with respect to your privacy, but to do whatever I can to get support.

 

Thanks again. Sorry to be a bother.

 

Regards,

Brian Page

rdww60, 2017-09-13:
And the FPVlab link for ImmersionRC support:

 

http://fpvlab.com/forums/showthread.php?48990-Sbus-on-ezuhf-500mw-tx <http://fpvlab.com/forums/showthread.php?48990-Sbus-on-ezuhf-500mw-tx&p=840212#post840212> &p=840212#post840212

kilrah, 2017-09-13:
FrSky and SBUS are unrelated. They don't use DSM protocol.

Anyway I still don't understand anything. DSM2 has NEVER sent more than 6 channels. Even checked the code for 2.1.6. It DISPLAYED 8 in the radio's settings, but that was just a display bug. 

So if you saw an actual difference - it's unrelated.

kilrah, 2017-09-13:
FrSky and SBUS are unrelated. They don't use DSM protocol.

Anyway I still don't understand anything. DSM2 has NEVER sent more than 6 channels. Even checked the code for 2.1.6. It DISPLAYED 8 in the radio's settings, but that was just a display bug. 

So if you saw an actual difference - it's unrelated. Also all the references I see about digital input on EzUHF are SBUS, not DSM. @schwabe where did you find about it?

I cleaned up the posts, try to avoid the quotes if you reply by email instead of doing so on github.

schwabe, 2017-09-13:
@rdww60 the whole dialog is public on github (https://github.com/opentx/opentx/issues/5204)

schwabe, 2017-09-13:
@rdww60 the whole dialog is public on github (https://github.com/opentx/opentx/issues/5204).

@kilrah https://www.immersionrc.com/ezuhf-firmware-v1-53-setup/ refrences DSM2/X input from Taranis.

@rdww60 If the ezuhf depends on the 100k 8e3 format rather than 1008e2 format then their serial input is really broken and they should fix it.

schwabe, 2017-09-13:
Actually the change was 125kN2 to 125kN1 but still.

Try this firmware on your X9D+ to see if that is really the change that upsets the EzUHF module: http://plai.de/.tmp/2017-09-13-x9d+-125kn2.bin

rdww60, 2017-09-14:
Received. Let me test and get back to you. Thank you.

rdww60, 2017-09-15:
[image]
[image]

rdww60, 2017-09-15:
Well as you expected, it did not work, but thank you for providing the test file.
Unfortunately, I've seen no response from ImmersionRC on their "support site". I read this to be typical, what a shame.
The symptom, is that the TX and RX won't bind after the update using SBUS "race" setup 1.5.3. The TX is the EzUHF JR Module, and the RX is either the 4CH Lite or Nano.
If you have any other ideas/thoughts on where to go from here to get this addressed, I'm all ears.
Everything I've seen to-date suggests it is on the EzUHF side of things, but I don't know enough about the implementation details.
I suppose I could try 2.1.9, to see it it still works there, but where would that take get me?
Still can't take advantage of the new features in 2.2.0 for the newer hardware (like the TrampHV).

schwabe, 2017-09-15:
The test file has the same output as 2.2.0 so it should work.  But none of your screenshots of version is from that test file as compile date should 2 days ago.

rdww60, 2017-09-15:
The .bin you provided was for 2.1.6? I did that wrong then. I'll check back later with update

schwabe, 2017-09-15:
sorry, the bin file should have the same output as 2.1 but it should up as 2.2.0 build in september on the about screen.

rdww60, 2017-09-16:
So my friend, I read and re-read your comments. My conclusion, was that the test bin file you gave me is a 2.2.x bin file, however the snapshot I sent did not reflect the build date. That was puzzling, so I downloaded the bin file again into a new directory. The size of the 2 are different, the smaller being the 2nd. download. I wrote the new file to the Taranis and when booting saw a nightly build warning I didn't see before, and also the build date is as you said.
.
Guess what? IT WORKS!!!!
.
This, I believe, could be one of the contributing factors to others not being able to get their setup.
.
I'm going to repeat the exercise to be sure, but the screen shots show it working. TX and RX bind, Flight Controller auto detects as before.

Last question: Will it be possible to roll this into a future release as a "knob" somewhere in the model setup do you think? I could verify again if you want, let me know.

ImmersionRC owes you guys BIG TIME.

THANK YOU!!
[image]
[image]
[image]
[image]

rdww60, 2017-09-16:
And.....it is repeatable. I down loaded a different bin file using the normal flow (2017-06-23)
[image]
[image]

. That fails binding and further the flight controller fails auto-detect.
Load the test bin file back in, and works again. 🥇

schwabe, 2017-09-16:
I would rather have Immersion RC have a statement or fix for that. (or even better implement parsing MUTI serial instead which would give them 16ch)

rdww60, 2017-09-18:
I just want to say thanks again for working with me to find the problem.
Updated at also at http://fpvlab.com/forums/showthread.php?48990-Sbus-on-ezuhf-500mw-tx&p=840338#post840338

rdww60, 2017-10-17:
Hi Schwabe, sorry to bother you again on this...
I'm trying to enable lua scripts and am having some issues, so I am verifying setup, etc.
Can you tell me if the test bin file you provided (http://plai.de/.tmp/2017-09-13-x9d+-125kn2.bin) was built with lua and sqt5font build options? Thanks in advance. Regards.

schwabe, 2017-10-17:
@rdww60 not with sqtfont as never included and no idea if it was with lua. It was a one off test version anyway.

dwfuks, 2017-10-26:
Hi Schwabe, sorry again ...)
Can you make same "125kn2" mod of 2.2 firmware for taranis qx7 users? For one time.  Because there is no 2.1 version fo qx7 and ImmersionRC still doesn't answer about this issue.
If you will make it, choose for mod firmware with "lua" and "massstorage" build options.
thanks a lot

schwabe, 2017-10-26:
@dwfuks I would rather not do that and go down that rabbit hole

strepto42, 2019-05-25:
I realise this is a well out of date thread dealing with well out of date hardware, but I'd like to get it working with my x-lite. I've been using the taranis version above for two years without any dramas but it's a big radio to take travelling.

I'm happy to attempt to check out and build my own copy of the firmware (I am a coder) but some pointers about where to start looking and how to go about this would be helpful. Even just some dot points about which branch of code I might need etc :)

Of course if someone did want to build me a flashable binary that'd be fine too ^_^

kilrah, 2019-05-26:
You don't need to build anything, SBUS output has been built in by default on all radios for ages.