opentx/opentx #5206: Jenkins and/or download down

scrouillou, 2017-09-11:
Hi,

Trying to gather information about OpenTx on mega 2650 board, I am unable to access several resources and links related to the following domains : 

http://downloads.open-tx.org/MEGA2560/, from http://www.open-tx.org/2015/04/18/M2560-DIY-radio-documentation

> Not Found
> The requested URL /MEGA2560/ was not found on this server.
> Apache/2.4.10 (Debian) Server at downloads.open-tx.org Port 80

http://jenkins.open-tx.org/

> DNS error, this website is unavailable

Are those servers still available ? Are the documentations available through other sources ?

Thanks in advance.

kilrah, 2017-09-11:
Here in the repo. https://github.com/opentx/opentx/tree/2.2/doc/Mega2560

closed 2017-09-11 by kilrah

scrouillou, 2017-09-11:
Thank you very much!