opentx/opentx #5209: External serial/usb control of Tx

aL3891, 2017-09-12:
Hello, first of all thanks for creating this project and all your hard work!

I wonder if its possible to control an OpenTX transmitter externally and specifically set channel values and get telemetry (mainly wondering about the first part). I've seen that there is some kind of serial support and a cli, but I have not found any reference for what the commands are but looking at those issues it seems mostly for accessing menus and debugging but those issues were quite old. 

Being able to control the transmitter from a PC would open up a lot of cool scenarios such as UAVs and ground stations in general. I'd love to take a stab at it myself but i'm not quite sure where to start :)

Sorry if this is a noob question, I've just discovered OpenTX and my TX is still in the mail.

vifino, 2017-09-15:
I also wondered about doing such a thing, it would for sure be a nice thing to have.

As far as I know, there is nothing like USB HID but in reverse, so it'd have to be a new and custom interface, which is a bit unfortunate. Protocol/Interface should be sufficiently flexible to maybe make it reusable?

schwabe, 2017-09-15:
@vifino better join in chat to discuss what you really need

projectkk2glider, 2017-09-15:
We talked about this before see #2604 , #2008  and #1159

aL3891, 2017-09-16:
We spoke a little about this in the chat, i haven't been able to look yet but it seems doable in some form, i was thinking adding another compile switch that uses the usb connection as a com port, just like cli, but instead accepts sbus as input and writes s.port as output. If that is possible i think it would be really cool because it would make a lot of other libraries "just work"

schwabe, 2017-09-16:
@aL3891 or just a cli command that is startcontrol and switch serial mode

closed 2019-11-15 by stale[bot]