opentx/opentx #5210: Is Horus X10S fully supported?

FieserKiller, 2017-09-12:
I've preordered a Horus X10S, frsky claims "OpenTX Support" but I don't find Horus X10S in the supported radios section of homepage and no mention in 2.2 release logs. 
I searched issues and checked the code, all I find are references to the other new radio Horus X10.
So how is the status of X10S? Will opentx 2.2 run on it? Or is it supported in nightlies already? Or is full support far, far away and I should better cancel my preorder?

kilrah, 2017-09-12:
They're the exact same thing, difference is only cosmetics.

closed 2017-09-12 by kilrah

FieserKiller, 2017-09-25:
Sorry for reopen, but "the internet" says horus x10s _is not_ supported. So could someone clarify that?

3djc, 2017-09-25:
Nor X10 or X10S are currently supported. They should be at some point in time, and no, there is not ETA for that

FieserKiller, 2017-09-25:
So could we reopen this issue then so people know?

3djc, 2017-09-25:
I do not see why, it is not a bug (tracking those is what this tool does)

FieserKiller, 2017-09-25:
So frsky is to blame for claiming opentx support while devs didn't know that? Ok, I'll get in touch with frsky

3djc, 2017-09-25:
The development of OpenTx is made by the community, not FrSky. It is based on unpaid, voluntary work, done by the devellopers during our spare time, in addition to our day to day job to feed the familly. We do our best to support the user community, but we do things with the time we have, not based on any coorporation own timing and agenda. Once again, we are planning to support those radio, but we haven't even seen a production version yet. So give us some time and we will do it

kilrah, 2017-09-25:
They're correct, OpenTX will run on it... when the work is done. I don't think anything implies it will be there on day one.

FieserKiller, 2017-09-25:
Guys, this goes really not against you, you are doing great work with opentx. 
But Frskys product page states clearly: "Comes with FrSky’s own easy to use FrOS (operating system) and of course, supports OpenTX as well." 
Source: https://www.frsky-rc.com/product/x10s/
I can't see how this could imply that opentx support is planned somewhere in the future. 
The first batch of radios is going to be shipped by end of this week, so next week first regular customers will receive their new baby and will try to install opentx to it. 
However, I contacted Frsky with a link to this issue, maybe some member of their staff will clarify their plans how they are going to assist you to make opentx support a reality.

schwabe, 2017-09-25:
We never promised any release date on OpenTX. And we are doing unpaid voluntary work in our spare time. We understand that it is frustrating for users not to have OpenTX on day one but as @3djc our own lifes and jobs come first.

kilrah, 2017-09-25:
> and will try to install opentx to it.

They won't since there is no version for them to try to put on.
Really just document yourself before your purchase, it's the least you could do. 
X12S also supports OpenTX, and the first official OpenTX release for it came >6 months after X12S availability.

> However, I contacted Frsky with a link to this issue, maybe some member of their staff will clarify their plans how they are going to assist you to make opentx support a reality.

There isn't anything that will help with, you're just stirring the pot and annoying people.

Hjeresen, 2017-11-29:
Any news about ETA on X10 Firmware?
I have checked everywhere!
Regards Bo

schwabe, 2017-11-29:
As already said, there are no ETAs. We are just volunteers.