opentx/opentx #5213: sdcard sync and hiden files

s0lstice, 2017-09-13:
Hello,

I'm under linux and during a synchronisation from local to sdcard I see the copy of ".trash/*" files.

Can we ignor the hiden files ?

You make a good work, thank you ;)

Mickaël

schwabe, 2017-09-13:
The Remote should not show them. Or do you mean during synchronisation?

s0lstice, 2017-09-14:
I can see hiden files on the remote (Horus). In "system/Card SD-HC" I have [.fseventsd], [.Spotlight-V100], [.Trashes], ... the other files are vilisble and needed.

As I deleted all local sdcard directory to use a new version without unsused files. The ".Trashes" dirctory contain all files of last version. In companion, the synchronization have teack a while.

I can't remove unecessary files because the sdcard is read only (#5118).

3djc, 2017-09-14:
Why is it mounted read only ???

s0lstice, 2017-09-14:
I don't know why. The horus firmware have the "mass storage" option activied. When I plug the horus I can't change the content of the sd card with the files browser. I need to use companion.

It is the same when I directly insert the sd card with an adapter.
I have a full access on another sd card mounted with the same adapter.

When I look what have done the automount with ":$mount" cli the answer seems to be good :

`:$ mount`
`/dev/sdc1 on /media/s0lstice/9016-4EF8 type vfat (rw,nosuid,nodev,relatime,uid=1000,gid=1000,fmask=0022,dmask=0022,codepage=437,iocharset=iso8859-1,shortname=mixed,showexec,utf8,flush,errors=remount-ro,uhelper=udisks2)`

3djc, 2017-09-14:
"It is the same when I directly insert the sd card with an adapter.
I have a full access on another sd card mounted with the same adapter."

Doesn't this says this as nothing to do with opentx ?

s0lstice, 2017-09-14:
I don't know enough opentx. This is my first radio with this software.

I can just say that I can not write on the sd card formatted for opentx from the computer. When I'm using an adapter or the USB from the radio. I need to use companion.

If my memory is good, in december, I have write on sd card to do the first install and the update few days after. This is the only SD card on which I can not write.

kilrah, 2017-09-14:
Then that SD is either faulty or needs reformatting. There's nothing specific to OpenTX with it, and companion uses nothing different from what your system would.

s0lstice, 2017-09-14:
I will format the sdcard and give you the resulat.

I'm working with the oepntx-x12s-2.2.0 (85863358), Date 2017-05-30, time 21:23:33.

It can be nice to see the hiden files on Horus and don't send it with companion. On scdard opentx don't need ".Trash", ".Sportlight-V100"; ...

schwabe, 2017-09-14:
OS X automatically creates those files. Not much we can do about that.

@kilrah Yes, we hide them.  (#4973)  But that commit was merged 2nd of June so slightly after 2.2.0 was released.

schwabe, 2017-09-14:
You can use a [snapshot build](http://www.open-tx.org/downloads) to see if files are hidden for you in a newer version.

hmichl, 2018-01-14:
I just configured sd-sync on a new installed open-tx 2.2.1. I use a mac and I still see files and directories starting with "." like .Trashes and .Spotlight-V100 synced between the sdcard and my sync-directory.
Shouldn't those hidden unix-files be igored already?

kilrah, 2018-01-14:
This is unrelated to sync. As soon as you connect the card to your Mac the finder will regenerate those on its own anyway.

There are tools to auto delete them...

notlebowski, 2018-06-16:
Can't the folders that start with a dot be ignored totally?
It is cluttering the progress screen and shows errors when in fact there are none.
AFAIK OpenTX does not use/create any folders that start with a dot.

A 2nd run should show no activity unless there is really something wrong.

The same goes for some Windows special folders like "System Volume Information"

mpaperno, 2018-06-17:
@notlebowski What are these errors?  Can it not read, or can it not write?

>AFAIK OpenTX does not use/create any folders that start with a dot.

Maybe (I'm not 100% sure), but files/folders starting with a dot are legitimate entries.... I'm not sure if always excluding them is such a good idea.  I'd prefer to use more definite criteria, is possible, or let the user decide.

> The same goes for some Windows special folders like "System Volume Information"

I'm surprised this isn't skipped already as it should be unreadable... can you post an example for this also?

Thanks.

mpaperno, 2018-06-17:
Never mind my last request about System Volume info.... found an example in #5735.

notlebowski, 2018-06-17:
Hi Max,

Thanks for looking into this.
Of course those entries starting with a dot are legitimate, but you have an IMHO too generic view of the sync tool within Companion.

You want to sync a folder with an SD-Card as good as possible. But OpenTX does not use files and folders that start with a dot.
When a folder or file starts with a dot it means it is supposed to be hidden. It has no use on the SD-Card and it would be better that those would not be created. The same goes for files/folders that have a hidden or system attribute on an NTFS-disk of the owner's computer.

Apple creates .spotlight folders that wind up on the SD-Card and when I sync it on my Windows machine these will land on my Windows system often with error messages because these different OS's are not mutual compatible with their files. Some file names that are valid on *Nix are invalid on Windows.

I have for instance a sync system (Syncthing) that keeps those folders on different computers in sync.

This software creates a system folder (starting with a dot) for its own housekeeping. They have it hidden and started with a dot in the hope that these are ignored in syncs. The companion sync tool destroys this hope. Optionally it has a revision system that keeps duplicates of changed and deleted files there. 

I want these on my computers, but not on my SD-Card.

OpenTX has no use for those and it not only clutters but also takes resources for nothing.

If the thing is optional it would be fine for me.

mpaperno, 2018-06-17:
Yea, look, we're dealing with at least 5 different file systems, each with their own idiosyncrasies and concepts of "hidden" and "system" files, and permissions (or lack thereof).  There are going to be compromises.  And don't get me started on OSX and its propensity for writing its housekeeping junk every time it touches a file system... :roll_eyes: 

I'm adding a filter option now, with default of skipping ".files" and NTFS/FAT hidden items. I think that will help and increase flexibility at the same time.  And if OTX ever starts using .files, don't come complaining to me.... :stuck_out_tongue: 

Now, can you tell me some of the errors in your case?  Is it always a write error, or are some of these files/folders also unreadable?  Are you always using SD sync on Windows/OSX, or both?  And is it always the hidden files/folders that cause errors (.files and/or with hidden flag on Windows) or do you get that with other items as well?

Thanks.

fraternl, 2018-06-17:
Indeed....
But my take on it is that you don't have to write a sync system that is used in a generic way.
It's only used for OpenTX and those hidden files/folders and files starting with a dot have no use there.

I'm indeed mixing OS's.
Currently I'm on holiday with only a MacBook Air and this system is creating .Spotlight folders which I don't want to be synced. Windows is creating a "System Volume Information" on the SD-Card and when the sync tool tries to create that on the OSX filesystem it can't and it's displayed in red.
Syncthing is an open-source cross-platform sync system that has its roots in Linux.
It therefore uses folders that start with a dot which it then excludes in its sync.
That folder should not land on another computer using that same sync system. It will if those are synced through the SD-Card.

Sometimes folders are created with deep structures that they will exceed limits of the Windows folders structure when they need to be placed in a subfolder. It's all one big mess which is not your fault, but you can be part of the solution if it just skips those.

It's for that reason I am asking you to make it optional, but I'm convinced I will tick it the moment you make it available and I will never need to untick it.

I can't see any reason why OpenTX should start using hidden folders and hidden files..
It's in part not even using long file names (yes I know that's not the filesystem).

mpaperno, 2018-06-17:
Yep, I get it... hidden and .files will be excluded by default in the next version of sync. :no_entry_sign:  :fist_right: :skull: :horse: 

In the meantime, just use Syncthing? I mean, what's the advantage of the Companion sync tool to begin with?  It doesn't even save your settings (yet). :)

A very simple batch or shell script could do any kind of custom sync.  "Even" Windows comes with a very powerful CLI sync tool (robocopy).

> Sometimes folders are created with deep structures that they will exceed limits of the Windows folders structure ... but you can be part of the solution if it just skips those.

Explain this please... created where?  [Here are the limits](https://msdn.microsoft.com/en-us/library/windows/desktop/ee681827%28v=vs.85%29.aspx?f=255&MSPPError=-2147217396#limits), are you exceeding 32K+ characters in a path name somehow?

notlebowski, 2018-06-17:
Syncthing is not managing the SD-Card at all. It just syncs the folders on the different computers.
The 3 radio's I have can be connected to either of the 4 computers I have Companion installed.
Syncthing keeps the computer folders in sync (3x Windows 10 and 1 MacBook Air).

Syncthing is not a replacement for the Companion sync tool. Your sync tool is the bridge to OpenTX itself. Syncthing is also creating folders that should not appear on the SD-Card nor on the other synced machines.

Syncthing will not sync the folders with a dot (an out of the box feature)
So the .Spotlight folders that are on the MacBook Air don't end up on the Windows machines.

Well... Until I use the Companion sync tool. The sync tool puts them on the SD-Card.
But luckily this stops now.

I'm a very happy user of the sync tool, but this was an ongoing annoyance.
I'm so glad that this is fixed now. It will also be much faster on top.

> Explain this please... created where? Here are the limits, are you exceeding 32K+ characters in a path name somehow?

If a deep path on the SD-Card has to be synced with a folder on the harddisk that's already in a deep path it has a good chance of being ignored.
Those .Spotlight folders and files can be very long.

mpaperno, 2018-06-17:
The Companion sync tool does absolutely nothing that is specific to OpenTX. It just copies files.

Another omission is that it doesn't remove "extra" files (which exist on destination but no longer on the source). That's a whole other ball of wax, which a "real" sync tool should already be able to handle with proper user options.

mpaperno, 2018-06-18:
I've been playing with sync and it already skips hidden files (flagged hidden on Windows and .files on Linux/OSX). And also Windows system files.  It skips over .DS_Store, System Volume Information, etc., altogether.  :confused: 

Is the problem that OSX isn't setting the hidden flag when creating files on FAT32?  So then CPN sync tries to copy them back because now they're not hidden?

Everything works seamlessly on Windows...  :grin:   even with old junk Mac journal files that I have scattered on the system drives and SD cards.  

I also don't understand why people have errors with System Volume Information, because, as I originally thought, it should already be ignored.  If we need to filter this by name, I wonder what else needs to be added to the list?

notlebowski, 2018-06-18:
This is all the junk that the synctool is copying even though it should have only copied tetris.luac from the X-Lite to the MacBook Air's hard drive.

The preceding dot is a *Nix way of excluding them from sight. I don't understand we can't agree that there is no place for them on OpenTX and Windows.

> The Companion sync tool does absolutely nothing that is specific to OpenTX. It just copies files.

That's exactly the problem I have with it. It SHOULD do things that are specific to OpenTX. This tool is not used with anything but Companion/OpenTX. Your approach to the tool (which I know you greatly enhanced, my thanks for that) is too generic.

Here's all the junk the tool is copying or trying to copy.
I DON'T WANT them to be copied at all. Just skip them. They have no use on the SD-Card and even less on my Windows system that later gets synced and wastes resources.
They also make a mess of the detailed view that is supposed to give me information about what really has changed between the SD-Card and the folder on my machine.
That 1 line is lost between all the crap that I don't want to be copied at all

The hidden folder with computer specific info for syncthing doesn't show up in this log, but it is handled by the synctool 

Nightly N636

There were some changes made by Kilrah yesterday.
Are they not working or are these not merged in N636??

```
Starting synchronization: /Users/MrWolf/SD.X-Lite -> /Volumes/SD22

Creating directory: /Volumes/SD22/.Spotlight-V100/Store-V2/A01932AE-D881-4AEF-80F3-98DC5A2DD155
Creating destination file: /Volumes/SD22/.Spotlight-V100/Store-V2/A01932AE-D881-4AEF-80F3-98DC5A2DD155/1.directoryStoreFile
Creating destination file: /Volumes/SD22/.Spotlight-V100/Store-V2/A01932AE-D881-4AEF-80F3-98DC5A2DD155/live.0.indexCompactDirectory
Creating destination file: /Volumes/SD22/.Spotlight-V100/Store-V2/A01932AE-D881-4AEF-80F3-98DC5A2DD155/4.indexPositions
Creating directory: /Volumes/SD22/.Spotlight-V100/Store-V2/A01932AE-D881-4AEF-80F3-98DC5A2DD155/journals.health_check
Creating destination file: /Volumes/SD22/.Spotlight-V100/Store-V2/A01932AE-D881-4AEF-80F3-98DC5A2DD155/0.shadowIndexHead
Creating destination file: /Volumes/SD22/.Spotlight-V100/Store-V2/A01932AE-D881-4AEF-80F3-98DC5A2DD155/3.indexPostings
Creating directory: /Volumes/SD22/.Spotlight-V100/Store-V2/A01932AE-D881-4AEF-80F3-98DC5A2DD155/journals.migration_secondchance
Creating destination file: /Volumes/SD22/.Spotlight-V100/Store-V2/A01932AE-D881-4AEF-80F3-98DC5A2DD155/1.indexUpdates
Creating destination file: /Volumes/SD22/.Spotlight-V100/Store-V2/A01932AE-D881-4AEF-80F3-98DC5A2DD155/3.indexPositions
Creating destination file: /Volumes/SD22/.Spotlight-V100/Store-V2/A01932AE-D881-4AEF-80F3-98DC5A2DD155/4.indexArrays
Creating directory: /Volumes/SD22/.Spotlight-V100/Store-V2/A01932AE-D881-4AEF-80F3-98DC5A2DD155/journals.assisted_import_pre
Creating destination file: /Volumes/SD22/.Spotlight-V100/Store-V2/A01932AE-D881-4AEF-80F3-98DC5A2DD155/2.directoryStoreFile.shadow
Creating destination file: /Volumes/SD22/.Spotlight-V100/Store-V2/A01932AE-D881-4AEF-80F3-98DC5A2DD155/0.indexPostings
Creating destination file: /Volumes/SD22/.Spotlight-V100/Store-V2/A01932AE-D881-4AEF-80F3-98DC5A2DD155/0.indexIds
Creating destination file: /Volumes/SD22/.Spotlight-V100/Store-V2/A01932AE-D881-4AEF-80F3-98DC5A2DD155/1.indexIds
Creating destination file: /Volumes/SD22/.Spotlight-V100/Store-V2/A01932AE-D881-4AEF-80F3-98DC5A2DD155/4.indexTermIds
Creating destination file: /Volumes/SD22/.Spotlight-V100/Store-V2/A01932AE-D881-4AEF-80F3-98DC5A2DD155/3.indexTermIds
Creating destination file: /Volumes/SD22/.Spotlight-V100/Store-V2/A01932AE-D881-4AEF-80F3-98DC5A2DD155/4.indexPostings
Creating destination file: /Volumes/SD22/.Spotlight-V100/Store-V2/A01932AE-D881-4AEF-80F3-98DC5A2DD155/live.0.directoryStoreFile.shadow
Creating destination file: /Volumes/SD22/.Spotlight-V100/Store-V2/A01932AE-D881-4AEF-80F3-98DC5A2DD155/2.indexPositions
Creating destination file: /Volumes/SD22/.Spotlight-V100/Store-V2/A01932AE-D881-4AEF-80F3-98DC5A2DD155/1.indexArrays
Creating destination file: /Volumes/SD22/.Spotlight-V100/Store-V2/A01932AE-D881-4AEF-80F3-98DC5A2DD155/3.indexGroups
Creating destination file: /Volumes/SD22/.Spotlight-V100/Store-V2/A01932AE-D881-4AEF-80F3-98DC5A2DD155/live.0.indexHead
Creating destination file: /Volumes/SD22/.Spotlight-V100/Store-V2/A01932AE-D881-4AEF-80F3-98DC5A2DD155/live.0.indexUpdates
Creating destination file: /Volumes/SD22/.Spotlight-V100/Store-V2/A01932AE-D881-4AEF-80F3-98DC5A2DD155/2.directoryStoreFile
Creating destination file: /Volumes/SD22/.Spotlight-V100/Store-V2/A01932AE-D881-4AEF-80F3-98DC5A2DD155/0.indexUpdates
Creating destination file: /Volumes/SD22/.Spotlight-V100/Store-V2/A01932AE-D881-4AEF-80F3-98DC5A2DD155/1.indexHead
Creating destination file: /Volumes/SD22/.Spotlight-V100/Store-V2/A01932AE-D881-4AEF-80F3-98DC5A2DD155/tmp.SnowLeopard
Creating destination file: /Volumes/SD22/.Spotlight-V100/Store-V2/A01932AE-D881-4AEF-80F3-98DC5A2DD155/journalExclusion
Creating destination file: /Volumes/SD22/.Spotlight-V100/Store-V2/A01932AE-D881-4AEF-80F3-98DC5A2DD155/2.indexBigDates
Creating destination file: /Volumes/SD22/.Spotlight-V100/Store-V2/A01932AE-D881-4AEF-80F3-98DC5A2DD155/live.0.indexPositionTable
Creating destination file: /Volumes/SD22/.Spotlight-V100/Store-V2/A01932AE-D881-4AEF-80F3-98DC5A2DD155/0.indexGroups
[... cut ...]
Replacing destination file: /Users/MrWolf/SD.X-Lite/.Spotlight-V100/Store-V2/655C89A1-E22C-4A4C-AAE0-D371F0A8F7BE/live.1.shadowIndexDirectory
Replacing destination file: /Users/MrWolf/SD.X-Lite/.Spotlight-V100/Store-V2/655C89A1-E22C-4A4C-AAE0-D371F0A8F7BE/live.1.directoryStoreFile.shadow
Replacing destination file: /Users/MrWolf/SD.X-Lite/.Spotlight-V100/Store-V2/655C89A1-E22C-4A4C-AAE0-D371F0A8F7BE/live.1.shadowIndexGroups

Finished synchronizing /Volumes/SD22 -> /Users/MrWolf/SD.X-Lite :
     New: 2; Updated: 35; Skipped: 1666; Errors: 3;
================================================================================

```

schwabe, 2018-06-18:
Calm down, @kilrah did not make any changes to syncing files in companion.

And for being too generic. Well the syn is doing what is supposed to do. Sync SD Card with a folder own your machine.

And we are looking into the problem, no need to get angy.

notlebowski, 2018-06-18:
I'm not angry at all. 
I'm giving feedback.
If it's because I'm using words like "crap"... It's just what it is to me in this context.

Tell me what makes you think that I'm angry. Maybe I can change my writing style so it better reflects my feelings

notlebowski, 2018-06-18:
> Calm down, @kilrah did not make any changes to syncing files in companion.
Ah.. Now noticed. It was a change in OpenTX so we don't see "System Volume Information" in the SD-Card manager. That's working!!

> And for being too generic. Well the syn is doing what is supposed to do. Sync SD Card with a folder own your machine.

You're giving me now the impression you don't want it to change. IMHO this is not the task of the sync tool. In my opinion it should sync the SD-Card with the a folder on my machine only files and folders that are relevant to OpenTX.

OS's and some applications (like Syncthing) are giving attributes to those folders and files that helps you hide them (preceding dot, hidden and system attributes). I think it's a shame that this is not being used as it now forces me to:

* Check a 3 pages log for maybe 1 little change
* Make sure those folders are removed before they are getting synced on other machines

schwabe, 2018-06-18:
@notlebowski writing whole sentences in UPPERCASE is considered screaming and I interpret screaming at me as being angry.

notlebowski, 2018-06-18:
I uppercased a few words to stress them. Not whole sentences.
Github doesn't use a wysiwyg editor so I did that instead of writing it in bold.
But I will leave that if it worries you

notlebowski, 2018-06-18:
> writing whole sentences in UPPERCASE is considered screaming and I interpret screaming at me as being angry.

In RFC documents they capitalize words like SHOULD and MUST in documents all the time. Exactly like I did (not whole sentences).
I'm convinced that those people weren't angry all this time.

https://tools.ietf.org/html/rfc2119

schwabe, 2018-06-18:
You asked why I get that impression and I told why got that impression. But consider that you want something from us. And perhaps you are better to adapt to the discussion style used here, instead of trying to lecture us on RFC document style (which I have almost seen being used in Github issues). 

To put it very blunt, it does not matter if you are right or not, if you piss us off, we are much less likely to invest our spare time in your problems and look into other things to do instead.

notlebowski, 2018-06-18:
I have a feeling you are angry.
Yet you didn't write in uppercase...

notlebowski, 2018-06-18:
I thought this ticket system was there to help you find the bugs in the system you create.
It seems it's not

kilrah, 2018-06-18:
Just stop it already. 
You did manage to calm down and keep civil for a while after your "identity change" which was fine, but if you're going to take your ridiculous hostile attitude again just stay away and refrain from "helping" us.

We are keen on improving our system, but not under pressure or aggression.

notlebowski, 2018-06-18:
Yes, you're right about the identity change.
The only reason I did that because there was no way back. Nothing I would write or do would come of anything.
I also stopped posting as I'm fully sincere in improving OpenTX.

Really read this ticket with objectivity and you would see there was no aggressiveness anywhere until Schwabe accused me of being angry.

kilrah, 2018-06-18:
There was a way back, and it was not to be obnoxious like you managed for a while. But now you got people angry again by keeping on repeating how bad it was. One is enough. 
Even after the change was accepted with @mpaperno saying 
>"Yep, I get it... hidden and .files will be excluded by default in the next version of sync." 

you kept bashing at the issue (and at us) for 3 more posts for absolutely no good reason, exactly like in the past.

As said we don't want any of that here. Either you're nice or you go away.

notlebowski, 2018-06-18:
I didn't keep "bashing at the issue".
This morning I tested the latest nightly and gave feedback it was not working as I would have hoped.

If it's my misunderstanding that I should wait longer to see the wanted change. Well, that's a mistake.

I merely gave the feedback Max was asking for. He still didn't get the log file where all those spotlight files were.

Then suddenly I was accused for being angry merely because I capitalized "should" and "don't want".

I don't think you have any reason to be offended by me. I don't even think Max is.

mpaperno, 2018-06-23:
Syncthing should be marking those folders as hidden instead of just with a . prefix.  This is a Unix convention, not Windows.  If Syncthing is meant to work with Windows then it should be putting the proper flags on whatever junk it creates in your folders.  (Is there REALLY no option for it to NOT create its housekeeping files inside the source/destination folder?  I can think of several situations where that might not even be possible or at least highly undesired.)

I do have proper exclusion filters working in a new sync tool, but it turned out to not be trivial (it's much easier to only include certain folders than exclude them).  Also discovered a couple Qt bugs along the way... joy.  (After that I developed a fever, but it's *probably* unrelated! :P )  This most likely won't make it into 2.2.2 since it's a significant rewrite.

> That 1 line is lost between all the crap that I don't want to be copied at all

Ironic, since all I was looking for was one line from your log, instead I got several pages of repeated "information" we've already covered, and then a bunch of bickering.  Generally, repeating a point (several times) after you've already made it is considered noise/unhelpful/perhaps rude. YMMV.

kilrah, 2018-06-23:
Syncthing does things right. My guess is that those files have been synced outside of it. Clean things up and it'll work.

notlebowski, 2018-06-28:
The /.sync/Archive was created by Syncthing.
It stores changed and removed files there.

I just checked these folders.
Although the /.sync folder also has a MS-Windows hidden flag, the /.sync/Archive folder does not.
Maybe the the folder /,sync/Archive is taken by the sync because it has no hidden attribute.

I can of course add a hidden attribute to it by hand on all my machines, but imho it should discard all subfolders beneath a hidden folder.
I'm satisfied as it is now. I don't think anyone besides me is now having any issue with it.
Thanks for all the changes.